moveit/moveit #578: [ikfast_template] searchPositionIK doesn't consider other solutions if the best solution returned by getPositionIK is in collision

hartmanndennis, 2017-08-08:
### Description

#537 fixed the ikfast plugin so that getPositionIK now returns the nearest solution to the seed state instead of the first solution returned by IKFast. 
Most moveit components call searchPositionIK which has an constraint callback parameter which getPositionIK lacks.
If the kinematics of your robot has no free parameters, searchPositionIK just calls getPositionIK and calls the constraint callback for this single solution. If the constraint fails, the function returns with no solution found even if there are other valid solutions which are not nearest to the seed.
To fix this searchPositionIK could instead call the getPositionIK version which returns every solution, sort them by distance and check the constraints in ascending order until it finds a solution which meet the constraints.

v4hn, 2017-08-08:
:+1: This sounds like a very good idea for ikfast.

nsnitish, 2017-08-09:
I think implementation will be more or less similar to #537

PR #585 (2017-08-19): IKFastTemplate: searchPositionIK now returns collision-free solution which is nearest to seed state.
Fixes #578
closed 2017-08-19 by v4hn