moveit/moveit #579: `Found empty JointState message` ROS_ERROR output with Kinetic

skohlbr, 2017-08-08:
### Description

It appears there is a regression in Kinetic that leads to `Found empty JointState message` ROS_ERROR output getting spammed to the terminal. See also ROS Answers: https://answers.ros.org/question/266274/moveit-regression-found-empty-jointstate-message
It appears for most (all?) people this does not have adverse effects. However, there is at least one regression related to joint states in Kinetic that might be related (#501).

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build
* 0.9.8-0xenial-20170628-083013-0800

### Steps to reproduce
Start robot in Gazebo simulation, use rviz Motion Planning plugin to plan and execute a motion.

### Expected behaviour
Planning should proceed without errors.

### Actual behaviour
Multiple ROS_ERROR messages are printed out as in the console output below.

### Backtrace or Console output

> [ INFO] [1502218727.220033766, 86.202000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
> [ INFO] [1502218727.220192486, 86.202000000]: Planning attempt 1 of at most 1
> [ERROR] [1502218727.220300333, 86.202000000]: Found empty JointState message
> [ERROR] [1502218727.220377697, 86.202000000]: Found empty JointState message
> [ERROR] [1502218727.223660471, 86.204000000]: Found empty JointState message
> [ERROR] [1502218727.229793354, 86.204000000]: Found empty JointState message
> [ INFO] [1502218727.230916230, 86.204000000]: Planner configuration 'arm_with_gripper_group' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.

skohlbr, 2017-08-08:
Following the ROS Answers link, this was likely introduced via #499.

Vonstafenberg, 2017-08-09:
I am having the  same problem....!

skohlbr, 2017-08-09:
@v4hn Given you appear pretty confident this is purely cosmetic in #499: Do you have an explanation why when planning is triggered, an empty `robot_state` is passed to [_robotStateMsgToRobotStateHelper](https://github.com/ros-planning/moveit/pull/499/files#diff-e03d4d985fd8f5e9565368bddcac8d96L329) (or at least one that contains empty name vectors in either of the two fields)?

v4hn, 2017-08-10:
I already tracked this down and have a possibly incomplete patch. Coming soon.
I'm on vacation though, so I don't want to spend the whole day in front of my screen. :)

closed 2017-08-14 by jrgnicho

ImSubin98, 2024-07-05:
I am having the same problem. Does anybody solve this problem?

PR #589 (2017-08-14): robotStateMsgToRobotState: is_diff==true => not empty
In case the incoming RobotStateMsg is a diff, the joint fields are allowed to be empty.
This is used to change attached objects and it's an intended use-pattern in MoveIt to modify the current state with some (possibly) incoming state (see e.g. [the fix_start_state_bounds adapter](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_request_adapter_plugins/src/fix_start_state_bounds.cpp#L85))

This needs to be cherry-picked indigo due to #518 .

Fixes #579