moveit/moveit #583: [MoveGroupInterface] setJointValueTarget doesn't check collisions during IK call

hartmanndennis, 2017-08-09:
setJointValueTarget adds an empty constraint callback to setFromIK as can be seen here:
https://github.com/ros-planning/moveit/blob/7d36403187dffbad7a747e65b98dd266cc0b9da9/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L463-L465
I don't think this is critical since the planner will check collisions later but in light of #578 it would be nice if a collision detection callback could be used.
Looking at MoveGroupCartesianPathService a planning_scene is needed to accomplish this and MoveGroupInterface currently doesn't need one. I never worked with planning scenes so I don't know how you would get one without much overhead and if it is even worth the effort.

v4hn, 2017-08-10:
You are right, this would be nice. But.

On Wed, Aug 09, 2017 at 02:54:59PM +0000, Dennis Hartmann wrote:
> Looking at MoveGroupCartesianPathService a planning_scene is needed to accomplish this and MoveGroupInterface currently doesn't need one.

Correct.

> I never worked with planning scenes so I don't know how you would get one without much overhead and if it is even worth the effort.

Short answer: in this case there isn't. We would have to maintain a `PlanningSceneMonitor` that adds a delay from `move_group`s scene
or invoke a service call request a current planning scene in each function call. Both solutions add too much complexity to
this simple interface. Maintaining an incomplete planning scene (only the robot) could be done, although it also adds some overhead,
but is broken by design in my opinion.

I would like to close this issue as "Won't fix".

hartmanndennis, 2017-08-10:
It's fine with me.

v4hn, 2017-08-10:
One thing we *could* do of course is to expose the callback parameter as another function parameter of `setJointValueTarget`.
This leaves the burden with the user and if they really want to, they can fetch a planning scene before. :-)

hartmanndennis, 2017-08-11:
That could be an option. Would it be better then to add the optional callback parameter after the optional end effector link?

v4hn, 2017-08-11:
We will not break API for this.
If you want to add it, either add a new default parameter at the end or
create a new method.