moveit/moveit #584: [indigo] ikfast plugin fails to compile with c++11 enabled

hartmanndennis, 2017-08-10:
#515 was cherry-picked to indigo but did someone test this on indigo? As far as I can see there are no tests for this and the template is never compiled so Travis doesn't fail.
Using c++11 fails for me on indigo with all updated 14.04 packages (with the indigo-release docker image as well) because of ambiguous isnan calls in the ikfast solver source. This is because of [this bug](https://sourceware.org/bugzilla/show_bug.cgi?id=19439) which is fixed in later libc versions but not in the one used in 14.04.
So I think this should be reverted in indigo and this would also prevent #575 because #537 added C++11 features.

gavanderhoorn, 2017-08-10:
I think we could also resolve the ambiguous `isnan(..)` calls (like was done in https://github.com/ros-industrial/motoman_experimental/commit/2981de24185b196fc3a345a96f273aba6fe85e38).

The template could be updated, and that should work on Indigo as well.

---

This was one of the reasons why I hesitated in #515 to cherry-pick this to Indigo.

hartmanndennis, 2017-08-10:
The template script would probably need to change all isnan calls to std::isnan like in your example. A simple `#undef isnan` sadly doesn't work.

gavanderhoorn, 2017-08-10:
`<ugly>`Could we not `#define isnan std::isnan` right before the `#include "_ROBOT_NAME___GROUP_NAME__ikfast_solver.cpp"`?`</ugly>`

hartmanndennis, 2017-08-10:
That could work. It does compile with this hack.

gavanderhoorn, 2017-08-10:
I've added some comments to your PR.

Thanks for reporting and submitting btw 👍

v4hn, 2017-08-14:
> As far as I can see there are no tests for this and the template is never compiled so Travis doesn't fail.

A test for this is very much welcome!

Thanks a lot for your work here. 👍

v4hn, 2017-09-26:
closing in favor of #619

closed 2017-09-26 by v4hn