moveit/moveit #587: createMeshFromResource crashes with inventor stl files

broesdecat, 2017-08-11:
### Description

When calling createMeshFromResource with an stl mesh exported from inventor, the program crashes on a memory corruption. Call stack below.
Workaround: open the inventor stl file in meshlab (no issue there) and export is as-is as an stl. Calling createMeshFromResource on the resulting stl file works fine.

The STL files are in attachment, both are binary.

### Your environment
* ROS Distro: Indigo
* OS Version: ros:indigo docker container (aka Ubuntu 14.04)
* Binary build 0.7.11-0trusty-20170621-105324-0800

### Backtrace or Console output
...
#5  0x00007fe89d6abae0 in __GI___libc_malloc (bytes=2920) at malloc.c:2893
#6  0x00007fe89dc66dad in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007fe89dc66ea9 in operator new[](unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
...
#10 0x00007fe89a4aa2c3 in Assimp::Importer::ReadFile(char const*, unsigned int) () from /usr/lib/libassimp.so.3
#11 0x00007fe89a4acfdd in Assimp::Importer::ReadFileFromMemory
#12 0x00007fe89e6bc235 in shapes::createMeshFromBinary
#13 0x00007fe89e6bc5d3 in shapes::createMeshFromResource
#14 0x000000000041cd42 in main ()
[mesh.zip](https://github.com/ros-planning/moveit/files/1218351/mesh.zip)

gavanderhoorn, 2017-08-11:
Could you add frames 8 and 9 as well to the bt?

broesdecat, 2017-08-11:
Updated the post.

Additional update: .obj files exported from Inventor are working fine.

gavanderhoorn, 2017-08-11:
Can you add what the actual crash is? Is it a `SEGFAULT` or stack / heap corruption or something else?

gavanderhoorn, 2017-08-11:
And just some expectation management: often times these things turn out to be problems in Assimp. As `geometric_shapes` is just a consumer of that code, we might not be able to do something about it.

---

Edit: thinking about it, perhaps it would be better to report this over at [ros-planning/geometric_shapes/issues](https://github.com/ros-planning/geometric_shapes/issues).

broesdecat, 2017-08-25:
Actual crash: `malloc(): memory corruption: 0x00000000013f5390`
No debug symbols on the Ubuntu servers.

I am indeed aware that the issue probably lies in assimp, but it makes more sense for that to be logged by you as direct consumer of their library I think. Anyway, I will relog on `geometric_shapes`.

closed 2017-08-25 by broesdecat