moveit/moveit #588: stack smash while planning with a collision object in the planning scene

dh562231640, 2017-08-11:
### Description

stack smashing detected, and move-group will died with exit code -6 It happens when add a collision object in the planning scene and try to plan with this collision object in the planning scene. And log file can not be found by the path it gives to me

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* Binary build verision 0.9.8-0xenial-20170628-083013-0800

### Steps to reproduce
add a collision object to a planning scene and drag the end effector on the object (crash it), then the rviz will died. or set some way points which has collision with the object and use move_group.CartecomputesianPath() and then move group will died.

### Expected behaviour
I think it should avoid such shutdown for move-group and rviz, It should raise error and fail planning.

### Actual behaviour
move-group or rviz shutdown

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

v4hn, 2017-08-12:
I can't reproduce this with my setup.

Please provide more details. Do you observe this with a local source-build too?

dh562231640, 2017-08-14:
@v4hn Thanks for the advice! I am trying to use source build one only to met a catkin_make error:

moveit/moveit_core/collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/build.make:110: recipe for target 'moveit/moveit_core/collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/src/collision_world_fcl.cpp.o' failed
make[2]: *** [moveit/moveit_core/collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/src/collision_world_fcl.cpp.o] Error 1
CMakeFiles/Makefile2:15921: recipe for target 'moveit/moveit_core/collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/all' failed
make[1]: *** [moveit/moveit_core/collision_detection_fcl/CMakeFiles/moveit_collision_detection_fcl.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

Do you know why this happens and how to get through this?

gavanderhoorn, 2017-08-14:
Please include more of the console output.

The part that you show here does not include the actual error, just the bit that says "something went wrong".

dh562231640, 2017-08-14:
Thank you for the quick reply but the outcome is really too much as I can only copy part of it:

/usr/include/fcl/traversal/traversal_node_octree.h:93:29: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void OcTreeDistance(const OcTree* tree1, const OcTree* tree2,
                             ^
/usr/include/fcl/traversal/traversal_node_octree.h:93:50: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void OcTreeDistance(const OcTree* tree1, const OcTree* tree2,
                                                  ^
/usr/include/fcl/traversal/traversal_node_octree.h:108:34: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void OcTreeMeshIntersect(const OcTree* tree1, const BVHModel<BV>* tree2,
                                  ^
/usr/include/fcl/traversal/traversal_node_octree.h:123:33: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void OcTreeMeshDistance(const OcTree* tree1, const BVHModel<BV>* tree2,
                                 ^
/usr/include/fcl/traversal/traversal_node_octree.h:138:61: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void MeshOcTreeIntersect(const BVHModel<BV>* tree1, const OcTree* tree2,
                                                             ^
/usr/include/fcl/traversal/traversal_node_octree.h:154:60: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void MeshOcTreeDistance(const BVHModel<BV>* tree1, const OcTree* tree2,
                                                            ^
/usr/include/fcl/traversal/traversal_node_octree.h:169:35: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void OcTreeShapeIntersect(const OcTree* tree, const S& s,
                                   ^
/usr/include/fcl/traversal/traversal_node_octree.h:189:47: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void ShapeOcTreeIntersect(const S& s, const OcTree* tree,
                                               ^
/usr/include/fcl/traversal/traversal_node_octree.h:208:34: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void OcTreeShapeDistance(const OcTree* tree, const S& s,
                                  ^
/usr/include/fcl/traversal/traversal_node_octree.h:225:46: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   void ShapeOcTreeDistance(const S& s, const OcTree* tree,
                                              ^
/usr/include/fcl/traversal/traversal_node_octree.h:243:41: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   bool OcTreeShapeDistanceRecurse(const OcTree* tree1, const OcTree::OcTreeNode* root1, const AABB& bv1,
                                         ^
/usr/include/fcl/traversal/traversal_node_octree.h:243:62: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   bool OcTreeShapeDistanceRecurse(const OcTree* tree1, const OcTree::OcTreeNode* root1, const AABB& bv1,
                                                              ^
/usr/include/fcl/traversal/traversal_node_octree.h:243:80: error: expected unqualified-id before ‘*’ token
   bool OcTreeShapeDistanceRecurse(const OcTree* tree1, const OcTree::OcTreeNode* root1, const AABB& bv1,
                                                                                ^
/usr/include/fcl/traversal/traversal_node_octree.h:243:80: error: expected ‘)’ before ‘*’ token
/usr/include/fcl/traversal/traversal_node_octree.h:243:80: error: expected initializer before ‘*’ token
/usr/include/fcl/traversal/traversal_node_octree.h:292:42: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   bool OcTreeShapeIntersectRecurse(const OcTree* tree1, const OcTree::OcTreeNode* root1, const AABB& bv1,
                                          ^
/usr/include/fcl/traversal/traversal_node_octree.h:292:63: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   bool OcTreeShapeIntersectRecurse(const OcTree* tree1, const OcTree::OcTreeNode* root1, const AABB& bv1,
                                                               ^
/usr/include/fcl/traversal/traversal_node_octree.h:292:81: error: expected unqualified-id before ‘*’ token
   bool OcTreeShapeIntersectRecurse(const OcTree* tree1, const OcTree::OcTreeNode* root1, const AABB& bv1,
                                                                                 ^
/usr/include/fcl/traversal/traversal_node_octree.h:292:81: error: expected ‘)’ before ‘*’ token
/usr/include/fcl/traversal/traversal_node_octree.h:292:81: error: expected initializer before ‘*’ token
In file included from /usr/include/fcl/traversal/traversal_node_setup.h:48:0,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/src/collision_world_fcl.cpp:40:
/usr/include/fcl/traversal/traversal_node_octree.h:445:40: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   bool OcTreeMeshDistanceRecurse(const OcTree* tree1, const OcTree::OcTreeNode* root1, const AABB& bv1,
                                        ^
/usr/include/fcl/traversal/traversal_node_octree.h:445:61: error: invalid use of template-name ‘fcl::OcTree’ without an argument list
   bool OcTreeMeshDistanceRecurse(const OcTree* tree1, const OcTree::OcTreeNode* root1, const AABB& bv1,
[... cut ...]
                                            ^
/usr/include/fcl/traversal/traversal_node_octree.h:235:61: error: request for member ‘getRootBV’ in ‘tree->’, which is of non-class type ‘const int’
     OcTreeShapeDistanceRecurse(tree, tree->getRoot(), tree->getRootBV(),
                                                             ^
/usr/include/fcl/traversal/traversal_node_octree.h: At global scope:
/usr/include/fcl/traversal/traversal_node_octree.h:238:3: error: expected unqualified-id at end of input
   }

gavanderhoorn, 2017-08-14:
With C/C++ errors: always scroll up to the first error that is reported and copy-paste that. The rest is most of the times not interesting.

dh562231640, 2017-08-14:
Below is the error from the very beginning
It seems that something bad happened in the library FCL but I just do not know how to deal with that.

In file included from /usr/include/fcl/ccd/motion_base.h:46:0,
                 from /usr/include/fcl/collision_object.h:45,
                 from /usr/include/fcl/broadphase/broadphase.h:43,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:43,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_robot_fcl.h:40,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_world_fcl.h:40,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/src/collision_world_fcl.cpp:37:
/usr/include/fcl/BV/RSS.h: In member function ‘fcl::FCL_REAL fcl::RSS::volume() const’:
/usr/include/fcl/BV/RSS.h:114:39: error: ‘template<class S> struct fcl::constants’ used without template parameters
     return (l[0] * l[1] * 2 * r + 4 * constants::pi * r * r * r);
                                       ^
In file included from /usr/local/include/fcl/geometry/collision_geometry.h:44:0,
                 from /usr/local/include/fcl/narrowphase/collision_object.h:43,
                 from /usr/local/include/fcl/broadphase/broadphase_collision_manager.h:44,
                 from /usr/local/include/fcl/broadphase/broadphase_bruteforce.h:42,
                 from /usr/include/fcl/broadphase/broadphase.h:233,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:43,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_robot_fcl.h:40,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_world_fcl.h:40,
                 from /home/dh/catkin_ws/src/moveit/moveit_core/collision_detection_fcl/src/collision_world_fcl.cpp:37:
/usr/local/include/fcl/math/bv/AABB.h: At global scope:
/usr/local/include/fcl/math/bv/AABB.h:49:7: error: ‘fcl::AABB’ is not a template type
 class AABB

gavanderhoorn, 2017-08-14:
Did you follow the [moveit installation instructions](http://moveit.ros.org/install/source/) for source builds?

Please provide some info on how you build moveit from source.

dh562231640, 2017-08-14:
I just did as it told me. nothing different

dh562231640, 2017-08-14:
I solved it!
it is the pcl that I installed from the source that replaced the old one and the downloaded one is not good for moveit and made the moveit runs bad.
Thank you for the helping

felixvd, 2018-09-06:
So this can be closed.

closed 2018-09-06 by rhaschke