moveit/moveit #590: No self collision during the initial Inverse Kinematics process of "computeCartesianPath" function

littlejohnyang, 2017-08-15:
### Description

I noticed when I am doing "computeCartesianPath", the joint trajectory solution includes poses with links colliding with itself. Can you confirm this? It seems to be doing collision detection with the 'world' properly. Can you confirm as well?

Is it possible to add the self-collision detection effectively? I do notice that all the self collision detection happens at "kinematic_constraints" class. Potentially I can add a binding to kinematic_constraints::VisibilityConstraint::decide() function. Is it a good idea?

### Your environment
* ROS Distro: Jade
* OS Version: Ubuntu 14.04
* Source build (can't remember the tag)

v4hn, 2017-08-16:
Sorry, but we recently froze the `jade-devel` branch. Can you reproduce this on kinetic or indigo?

littlejohnyang, 2017-08-17:
I spent a bit of time digging into the source code now. I can confirm that the self collision function is definitely being called. However the collision happened only between some of the links but was fine on other pairs of links. 

So I created a separate package with my own node and extracted out the relevant computeCartesianPath code out from MoveIt source code. The computeCartesianPath is now linked with the constraint function checking the collision, shown below:
```
bool isStateValid(const planning_scene::PlanningScene *planning_scene,
                  const kinematic_constraints::KinematicConstraintSet *constraint_set,
                  robot_state::RobotState *state,
                  const robot_state::JointModelGroup *group, const double *ik_solution)
{
  state->setJointGroupPositions(group, ik_solution);
  state->update();
  return (!planning_scene || !planning_scene->isStateColliding(*state, group->getName())) &&
    (!constraint_set || constraint_set->decide(*state).satisfied);
}
```
The computeCartesianPath function call:
```
constraint_fn = boost::bind(&isStateValid, static_cast<const planning_scene::PlanningSceneConstPtr&>(*ls).get() , kset->empty() ? NULL : kset.get(), _1, _2, _3);

double fraction = start_state.computeCartesianPath(jmg, traj, start_state.getLinkModel(link_name), waypoint3D, global_frame, step_size, jump_threshold, constraint_fn);
```
I have a 7 DOF arm. I tried the following code on collision detection:
```
// Test collision detection
    double ik_solution[7]; // selected pose
    ik_solution[0]=2.44;
    ik_solution[1]=-0.86;
    ik_solution[2]=1.35;
    ik_solution[3]=2.58;
    ik_solution[4]=-3.14;
    ik_solution[5]=1.07;
    ik_solution[6]=0.97;

    start_state.setJointGroupPositions(jmg, ik_solution);
    start_state.update();
    bool collision = planning_scene->isStateColliding(start_state, jmg->getName());
    ROS_INFO_STREAM("cartesianPathPlan: collision="<<collision);
```
When I picked a pose so that link7 and link4 is in collision, the above code will return (true) as it is colliding. However when I do it with link7 and link5, it just won't pick it up. I verify the collision visualisation by using gazebo, rviz and rqt. The robot_description file is correct so I believe the TF is calculated correctly. I also checked ACM so they are definitely not ignoring collision check between these links.

So I am suspecting few things now:
1) Can the dae model file for this particular link be loaded incorrectly so the FCL collision check won't perform as expected?
2) TF transforms is incorrect for the links when doing collision checking?

littlejohnyang, 2017-08-17:
Instead of using the complex dae files, I recreated the model using box objects for collision avoidance, it now works properly. So it should be just how the dae model is converted to collision obj in the code.

gavanderhoorn, 2017-08-17:
Using Collada for collision meshes used to be a problem (ros-planning/moveit#159), but with the merge of ros-planning/geometric_shapes#52 the idea was that it should work again.

I'm not sure that what you're seeing is another manifestation / related, but I thought I'd mention it in any case.

> Instead of using the complex dae files, I recreated the model using box objects for collision avoidance, it now works properly.

And as a side note: using 'lower quality' (ie: lower face / vertex count) meshes for collision geometry is a good thing to do in any case, as it can greatly reduce planning time.

v4hn, 2017-09-11:
@littlejohnyang did you dig deeper into this?
I'm not sure what we can to do about this issue at the moment...

littlejohnyang, 2017-09-23:
I had a closer look at the dae files and I'm certain all the "triangles" of the mesh object has the normal facing the right direction. The dae files are generated directly from solidwork model which is a lot more complex and include things like screw holes etc. So if the collision detection is definitely not reporting correctly when using these complex models. The collision detection do report correctly if the two links have a larger area of overlap but not when some of the outer surfaces are touching each other. I suspect that's probably related to the how triangles are being interpreted when using AABB calculation. 

For my project, I am just going forward with "box" representation which works fine for me. It also speeds up the computation anyways.

v4hn, 2017-11-17:
Ok, but this does not leave us a way to debug this.

Closing until further evidents, and a reproducible minimal test-case appears.

closed 2017-11-17 by v4hn