moveit/moveit #591: planning_scene.distanceToCollision( robot_state ) returns a max value

AndyZe, 2017-08-15:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Binary version 0.7.12-0trus

### Steps to reproduce
Minimal example here: https://github.com/UTNuclearRoboticsPublic/standalone_collision_check

### Expected behaviour
I spawn a collision cube then try to measure the distance with planning_scene.distanceToCollision(). It should be return something like ~0.5m.

### Actual behaviour
The returned value is 1.78e308

### Backtrace or Console output

### Comments
I'm fairly certain the cube is published correctly because MoveIt plans around it.

AndyZe, 2017-08-15:
Just checked ... same issue on Kinetic.

AndyZe, 2017-08-15:
I did eventually get it working per https://github.com/ros-planning/moveit/issues/430. I guess I was mixing syntax. The functional code:

`  ros::init (argc, argv, "collision_check");
  ros::NodeHandle nh;

  ros::Duration(1.0).sleep();

  // Set up the MoveIt scene
  robot_model_loader::RobotModelLoader robot_model_loader("robot_description");
  robot_model::RobotModelPtr kinematic_model = robot_model_loader.getModel();
  planning_scene::PlanningScene planning_scene(kinematic_model);
  robot_state::RobotState robot_state = planning_scene.getCurrentState();
  collision_detection::AllowedCollisionMatrix acm = planning_scene.getAllowedCollisionMatrix();

  // Spawn a virtual collision object (for testing)
  spawn_collision_cube(nh);

  // Spin while checking minimum collision distance
  while ( ros::ok() )
  {
    //process collision objects in scene
    moveit::planning_interface::PlanningSceneInterface planning_scene_interface_;
    std::map<std::string, moveit_msgs::CollisionObject> c_objects_map = planning_scene_interface_.getObjects(); //error here
    for(auto& kv : c_objects_map){
      planning_scene.processCollisionObjectMsg(kv.second);
    }

  //print distance
  ROS_INFO_STREAM("Distance to Collision: " << planning_scene.distanceToCollision(robot_state));

    ros::Duration(1.).sleep();
  }`

closed 2017-08-15 by AndyZe

v4hn, 2017-08-16:
```
  // Spin while checking minimum collision distance
  while ( ros::ok() )
  {
    //process collision objects in scene
    moveit::planning_interface::PlanningSceneInterface planning_scene_interface_;
    std::map<std::string, moveit_msgs::CollisionObject> c_objects_map = planning_scene_interface_.getObjects(); //error here
    for(auto& kv : c_objects_map){
      planning_scene.processCollisionObjectMsg(kv.second);
    }
```

Congratulations on getting it to work.

Please note that it would be better to work with the `/get_planning_scene` service (for which there is no generic wrapper function in the `PlanningSceneInterface` yet, patches welcome) and update the relevant part of the planning scene; alternatively you could create a `PlanningSceneMonitor` and have it subscribe to `move_group/monitored_planning_scene` to receive `move_group`'s PlanningScene with a small delay.

The code above will never delete a collision object from the planning scene - `getObjects()` never returns `REMOVE` operations.