moveit/moveit #597: Robot Model from URDF/SRDF String and Kinematic Plugins

Jmeyer1292, 2017-08-21:
### Description

Just a quick note:

With the current design of `robot_model_loader` and its helper class `RDFLoader`, one cannot load robot models from a urdf and srdf string, and then load the kinematics plugins from parameters. 

This is because the `RDFLoader` constructor using the strings never sets the `robot_description_` parameter of the class. This field is used later to create the kinematics plugin loader, regardless of how one initializes `RobotModelLoader`. The result is that ROS is looking up strings like `_kinematics/manipulator/` instead of `robot_description_kinematics/manipulator`. The result is that the loader silently fails to load any kinematic plugins.

Now this behaviour may be intended. I have a bit of an odd case where I wish to load multiple robots (with known group names) from string descriptions, but I want to use a standard set of kinematic plugins for them.

Mostly I wanted to post this for anyone else doing non-standard things: you need to create your own kinematics plugin loader and manage its life time. One easy change might be to use the options robot_description field instead of the rdf_loader one if the latter is empty but the former is not. 

### Your environment
* ROS Distro: Indigo
* OS Version:Ubuntu 14.04
* Source build

v4hn, 2017-09-06:
Sorry for the long delay! Apparently I never actually sent the mail.

A patch is very welcome. :)

At first glance it shouldn't be too hard to add support for this in the `RobotModelLoader`.

bmagyar, 2017-09-06:
And if you could add a test to nail the issue that would be even better :+1: