moveit/moveit #600: Model flash after trajectory

XuxinDoubleWang, 2017-08-25:
Hello everyone,
I am trying to use moveit to do inverse kin. 
I follow moveit tutorial from website and my arm controller is arbotix.
No bug and error when I launch my bringup launch file and moveit_group.
However, after I run final execution node. My model flash between final pose and initial pose like following: 
 
[image]
[image]

So, when I got the joint position, I always got the initial joint position.
In my opinion, there is some thing publish the initial position to lead to conflict. However, I cannot find it. 

My bringup launch file is shown below:

[image]

 
My excitation node is shown below:

`#!/usr/bin/env python

import math
import rospy, sys
import moveit_commander
from moveit_msgs.msg import RobotTrajectory
from trajectory_msgs.msg import JointTrajectoryPoint

from geometry_msgs.msg import PoseStamped, Pose
from tf.transformations import euler_from_quaternion, quaternion_from_euler

class MoveItDemo:
    def __init__(self):
        # Initialize the move_group API
        moveit_commander.roscpp_initialize(sys.argv)
        
        rospy.init_node('moveit_demo')
        

	robot = moveit_commander.RobotCommander()       
        # Initialize the move group for the right arm
        right_arm = moveit_commander.MoveGroupCommander('manipulator')
                
        # Get the name of the end-effector link
        end_effector_link = right_arm.get_end_effector_link()
                        
        # Set the reference frame for pose targets
        reference_frame = 'world'
        
        # Set the right arm reference frame accordingly
        right_arm.set_pose_reference_frame(reference_frame)
                
        # Allow replanning to increase the odds of a solution
        right_arm.allow_replanning(True)
        
        # Allow some leeway in position (meters) and orientation (radians)
        right_arm.set_goal_position_tolerance(0.01)
        right_arm.set_goal_orientation_tolerance(0.05)
        

        right_arm.set_start_state_to_current_state()
	right_arm.go()
        rospy.sleep(2)

        target_pose_start = PoseStamped()
        target_pose_start.header.frame_id = reference_frame
        target_pose_start.header.stamp = rospy.Time.now()     
        target_pose_start.pose.position.x = 0.05
        target_pose_start.pose.position.y = 0.05
        target_pose_start.pose.position.z = 0.60

	X = 0
	Y = 0
	Z = 0
	
        target_pose_start.pose.orientation.x = math.sin(Y/2)*math.sin(Z/2)*math.cos(X/2) + math.cos(Y/2)*math.cos(Z/2)*math.sin(X/2)
        target_pose_start.pose.orientation.y = math.sin(Y/2)*math.cos(Z/2)*math.cos(X/2) + math.cos(Y/2)*math.sin(Z/2)*math.sin(X/2)
        target_pose_start.pose.orientation.z = math.cos(Y/2)*math.sin(Z/2)*math.cos(X/2) - math.sin(Y/2)*math.cos(Z/2)*math.sin(X/2)
        target_pose_start.pose.orientation.w = math.cos(Y/2)*math.cos(Z/2)*math.cos(X/2) - math.sin(Y/2)*math.sin(Z/2)*math.sin(X/2)
        
        # Set the start state to the current state
        right_arm.set_start_state_to_current_state()
        
        # Set the goal pose of the end effector to the stored pose
        right_arm.set_pose_target(target_pose_start, end_effector_link)
        
        # Plan the trajectory to the goal
        traj = right_arm.plan()
        
        # Execute the planned trajectory
        right_arm.execute(traj)
    
        # Pause for a second
        rospy.sleep(1)
        

	print "============ Printing robot state"
	print robot.get_current_state()
	print "============"

        moveit_commander.roscpp_shutdown()
        moveit_commander.os._exit(0)

if __name__ == "__main__":
    MoveItDemo()

`

Does there someone know what the problem is and how to fix it?
Any ideas is very appreciated!!!

Thanks in advance.

* ROS Distro: [Indigo]
* OS Version: Ubuntu 14.04

v4hn, 2017-08-25:
Please never post screenshots of source code in bug reports..

The node that publishes joint states is probably the `joint_state_publisher` that simulates the behavior of the robot for you when you use `demo.launch`.

If you plug in another simulator the two will overwrite each other.

XuxinDoubleWang, 2017-08-26:
Sorry for the screenshot.
Thanks for your helpful idea!!
What do u mean " another simulator"? Could you please give me more details about it?

v4hn, 2017-08-27:
From the launch-file you screenshot the `arbotix_driver` node starts up in "fake mode".
Without investigating, I believe this is a mode similar to moveit's `demo.launch`.
So the arbotix_driver and the joint_state_publisher overwrite each other.

It's probably a good idea to remove the `joint_state_publisher` in this case.

Use `rqt_graph` to get an overview over the node network or `rostopic info /joint_states` to see which nodes publish joint state information.

closed 2017-08-30 by XuxinDoubleWang