moveit/moveit #605: Occurrence of "Invalid Trajectory: start point deviates"

crvogt, 2017-08-30:
Hi there,
This issue is coming up for me. I've done a clean ROS install, Indigo on Ubuntu 14.04, but it's continued to persist. I followed the instructions in this link http://moveit.ros.org/moveit!/ros/2017/01/03/firstIndigoRelease.html (I've both built it from source and tried the binaries. I've checked #290 and #292, as well as checked the actual changes made to the file to ensure I have built the most recent version with the currentPose - trajectoryPose line.

The error occurs consistently when running a simple node and the pr2_moveit_config demo.launch file.

Is it that moveit-full is still not offered, and that by downloading it I still have the old code with this bug still?

Any help is much appreciated :).

* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Source and Binary build
* Most recent source and binary

Steps to reproduce
roslaunch pr2_moveit_config demo.launch

rosrun (the following code sloppily adapted from the tutorials)
<script src="https://gist.github.com/crvogt/d41c55b5ebf240f7a81de43d487167e9.js"></script>

Error occurs:
<script src="https://gist.github.com/crvogt/4cfad1dec6a9fee4449b8b9768493f12.js"></script>

v4hn, 2017-08-31:
Your gist urls in this issue are wrongly formatted.

I see that between planning and execution you sleep a rather long time (10 seconds).
Usually physics simulations are not dead accurate over a long periode of time,
although I know that the PR2 arms might be somewhat of an exception there...

Does the problem persist if you *don't* sleep there at all / use the `move()` method
instead of plan&execute?

crvogt, 2017-08-31:
Ah sorry, I tried to post the code correctly (ie not just copy and paste), not sure how to do it with the gist urls...

Anyhow, I changed the sleep time, and the error does not occur on the first run through after catkin_make, but does every time after that :/

On replacing plan and execute, the error has not appeared, and the movements have been carried out as I'd hoped.

Soo, sleep() between plan and execute doesn't seem to be a great idea, I assume there's a drift/noise built in that would trip the error if there's too much sleep time? Though, too little sleep time cause errors as well, so the plan&execute configuration doesn't seem to be a great way to go. Am I too far off? 
(Either way you slice it, thanks for your help!)

v4hn, 2017-08-31:
replacing `plan;sleep;execute` by `move` should be basically equivalent to removing the sleep entirely w.r.t. the error you see.

> there's a drift/noise built in that would trip the error if there's too much sleep time?

This seems to be the case, yes.

> Though, too little sleep time cause errors as well,

This on the other hand doesn't make any sense at all to me at the moment.

> so the plan&execute configuration doesn't seem to be a great way to go.

If your controller can deal with a slight discrepancy between its current state and the start state of the trajectory
then you can relax the validation threshold instead. `0.01` was chosen as a rather strict threshold.

crvogt, 2017-08-31:
>Though, too little sleep time cause errors as well
>This on the other hand doesn't make any sense at all to me at the moment.

Ah, sorry, yes, that's caused a different error around when I attempt to attach an object to the gripper. Since I did a single variable for sleep time and decreased it to avoid the joint error, it doesn't seem to give enough time to attach an object to the robot...

>If your controller can deal with a slight discrepancy between its current state and the start state of the trajectory
then you can relax the validation threshold instead. `0.01` was chosen as a rather strict threshold.

Oh ok, it doesn't look like I'll need to do that for now, but it's good to know.

Thank you again for your help, it's definitely helped to clarify some of this, certainly enough to continue what I was doing!

closed 2017-08-31 by crvogt

v4hn, 2017-08-31:
On Thu, Aug 31, 2017 at 01:11:06PM +0000, crvogt wrote:
> it doesn't seem to give enough time to attach an object to the robot...

Consider using the synchronous `PlanningSceneInterface::applyAttachedCollisionObject` method instead, to avoid the (ridiculous..) sleep.
Due to backward compatibility (and complains from other maintainers...) we did not change the
attach method in the `MoveGroup` to use the service...

Good luck with your project.

crvogt, 2017-09-01:
:) Thank you for the tip! I've checked it out and will apply it.