moveit/moveit #608: Typo?

gerac83, 2017-09-04:
I found a typo as below:

https://github.com/ros-planning/moveit/blob/4bc43c56ca2a5293f3ef8d55796ec586cfde5f69/moveit_planners/ompl/ompl_interface/src/detail/ompl_console.cpp#L40

it should be ompl_interface. I have no idea if this is relevant at all.

bmagyar, 2017-09-04:
> I found a typo as below:
>
> https://github.com/ros-planning/moveit/blob/4bc43c56ca2a5293f3ef8d55796ec5
> 86cfde5f69/moveit_planners/ompl/ompl_interface/src/
> detail/ompl_console.cpp#L40
>
> it should be ompl_interface. I have no idea if this is relevant at all.

Thanks,  nice catch!  I'll look further into this today.

bmagyar, 2017-09-04:
I couldn't find any traces in the current moveit codebase of using this file at all.
As it does nothing but prints, I think it may even be time for this file to be retired...

v4hn, 2017-09-04:
If it is not referenced +1 to drop it.

bmagyar, 2017-09-05:
I'll sort that out :+1:

simonschmeisser, 2017-09-10:
can be closed?

bmagyar, 2017-09-10:
Yeah, thanks!

closed 2017-09-10 by bmagyar