moveit/moveit #611: python code is removed from warehouse_ros, but still exists in setup assistant

534o, 2017-09-11:
warehouse_ros python code is removed in jade and kinetic, as discussed in 
https://github.com/ros-planning/warehouse_ros/issues/35, https://github.com/ros-planning/warehouse_ros/pull/32

but still exists in https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/warehouse.launch#L10-L13

we need consistency between these two packages. keep python code on warehouse_ros or update moveit_setup_assistant package

v4hn, 2017-09-11:
The package that is referred to in this launch file is https://github.com/ros-planning/warehouse_ros_mongo
which indeed provides the `mongo_wrapper_ros.py` script.

Afair - but no guarantees - we could not release this repo via rosdistro because it depends on a mongo driver that is not packaged in ubuntu. I guess you didn't build it yourself in the CI environment.
See [here](http://moveit.ros.org/install/source/dependencies/) for build instructions.

Of course we could comment the node statement in this launch file, but from my point of view there is not much wrong with it at the moment.

What would you propose to do?

k-okada, 2017-09-19:
I had same problem and the problem for me, is when we use `roslaunch_add_file_check(launch)`(http://wiki.ros.org/roslaunch#Catkin ) on our cmake file, it marked as 'failure' because of missing `mongo_wrapper_ros.py` node. 

So my proposal is either
- release both mongo driver and warehouse_ros_mongo as ros package
- add mongo_wrapper_ros.py to wherehouse_ros and default user use that old script file. Only advanced developer who build mongo driver from source will use mongo_wrapper
   - BTW does mongo_wrapper_ros.py in wharehouse_ros_mongo depends on mongo c++ driver? it seems it just run `mongod` daemon... so we just put this script back to wharehouse_ros is ok.

I know this is very trivial thing, but for me MoveIt is very useful and fundamental software, so I'm expecting the code generated by moveit assistant is fully supported by default ros/ubuntu systems.

v4hn, 2017-09-19:
> release both mongo driver and warehouse_ros_mongo as ros package

@130s Could you have a look into this possibility?

Alternatively it would be best to work towards resolving #123.

> BTW does mongo_wrapper_ros.py in wharehouse_ros_mongo depends on mongo c++ driver? it seems it just run mongod daemon... so we just put this script back to wharehouse_ros is ok.

But this doesn't make much sense in my opinion. The point of the `warehouse_ros` refactoring was to get away from the hard dependency on mongodb.

My preferred solution until #123 is fixed or `warehouse_ros_mongo` is released would be to comment the lines from the launch file in the template.
If someone wants to make use of them, he needs to build the client side anyway.

130s, 2017-09-22:
>>  release both mongo driver and warehouse_ros_mongo as ros package
>
> @130s Could you have a look into this possibility?

Not entirely sure but with https://github.com/ros-planning/warehouse_ros_mongo/pull/8, the driver issue for jade branch seems to be addressed.

@davetcoleman looks like you have done some work there so would you share some insights?

130s, 2022-09-07:
>>>   release both mongo driver and warehouse_ros_mongo as ros package

It is an deb package since a while ago, maybe since https://github.com/ros/rosdistro/pull/22063, for ROS Melodic.

rhaschke, 2022-09-07:
Yes, that's correct. The `mongo_wrapper_ros.py` script which is part of [`warehouse_ros_mongo`](https://github.com/ros-planning/warehouse_ros_mongo) is functional and referred from MSA. All fine. I think we can close this issue.

closed 2022-09-07 by rhaschke