moveit/moveit #615: Integrating BITstar from OMPL to moveit

ljklonepiece, 2017-09-20:
### Description

I am trying to add BITstar motion planner algorithm from OMPL into moveit_planner

### Your environment
* ROS Distro: [Indigo]
* OS Version: Ubuntu 14.04
* Source: 
Moveit: https://raw.githubusercontent.com/ros-planning/moveit/indigo-devel/moveit.rosinstall

### Steps to reproduce
* set up moveit pkg
1. create a catkin workspace and go into src folder
2. `wstool init .`
3. `wstool merge https://raw.githubusercontent.com/ros-planning/moveit/indigo-devel/moveit.rosinstall`
4. `wstool update`
5. `rosdep install -y --from-paths . --ignore-src --rosdistro indigo`
6. `cd .. `
7. `catkin config --extend /opt/ros/indigo --cmake-args -DCMAKE_BUILD_TYPE=Release`
8. `sudo apt-get -qq remove ros-indigo-ompl`

* set up ompl pkg (in the same src folder)
1. `git clone https://github.com/ompl/ompl`
2. `cd ompl`
3. `git checkout 1.1.1` (this version can be compiled successfully)
4. `wget https://raw.githubusercontent.com/ros-gbp/ompl-release/debian/indigo/trusty/ompl/package.xml`
5. `cd ../..`
6. `catkin build`

* modify moveit_planner to add BITstar planner
1. modify src/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp. Include the header file that defines your planner at the top and register your planner in the registerDefaultPlanners()
2. modify /config/ompl_planning.yaml in fetch_moveit_config to include the option for the new planner.

* install fetch ros
1. `git clone https://github.com/fetchrobotics/fetch_ros.git`
2. catkin build again

* run fetch move_group.launch
1. roslaunch fetch_moveit_config move_group.launch
2. set start and goal pose in rviz
3. choose [BITstar planner name you defined in ompl_planning.yaml] 
4. click `Plan`

### Expected behaviour
plan a pathway and show the planned path in rviz

### Actual behaviour
moveit process died

### Backtrace or Console output 
[GDB backtrack msgs](https://gist.github.com/ljklonepiece/6357a347d3ce3c5dde71375560318900)

v4hn, 2017-09-20:
I didn't look into your issue, but did you have a look at #117 ?
Maybe you made some mistake during setup that the new ompl-planner-wrapper does better?

BryceStevenWilley, 2018-04-27:
I actually made a fix to support this, just saw this issue.

The key thing is that BIT* can only handle certain types of state spaces, since it needs to do informed sampling (this shows up as `PathLengthDirectInfSampler only supports RealVector, SE2 and SE3 StateSpaces.` in your error). However, MoveIt's model-based-state-space doesn't inherit from any of those state spaces. In theory, MBSS is exactly a RealVector state space, as it just puts the values in a [double array](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/model_based_state_space.h#L163), so the changes are pretty easy.

I made these changes in our own moveit_ompl plugin [here](https://github.com/KavrakiLab/moveit_ompl_planning_interface/commit/cc1517c26dd79c9cd24faba5865ff627f8818bc5) if you want to see how it's done. While it would be nice to have a PR for this in vanilla MoveIt, I think it'd be better to complete #117, as the plugin already has these changes.

davetcoleman, 2018-05-14:
As @BryceStevenWilley  said, the key issue seems to be:

>   what():  PathLengthDirectInfSampler only supports RealVector, SE2 and SE3 StateSpaces.

ascane, 2021-08-28:
I am also trying to add BITstar into moveit. Has anyone managed to do this already?