moveit/moveit #616: moveit setup show no active joints or end effectors found for group ''

jacknlliu, 2017-09-21:
I forked ROS packages from  [ROS-I/universal_robot](https://github.com/ros-industrial/universal_robot/). And I follow [Moveit Setup Assistant Tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html). After setup Moveit config, I got a `demo.launch` file, then   
```
roslaunch  ur5_grasp_moveit_config demo.launch
```

I got the following errors, and the `rviz` did not show any `interactive markers`.  

```
[ INFO] [1505957647.779692210]: Fake controller 'fake_ur5manipulator_controller' with joints [ shoulder_pan_joint shoulder_lift_joint elbow_joint wrist_1_joint wrist_2_joint wrist_3_joint ]  
[ INFO] [1505957647.788025775]: Fake controller 'fake_gripper_controller' with joints [ bh_j11_joint bh_j12_joint bh_j22_joint bh_j32_joint ]  
[ INFO] [1505957647.796457498]: Fake controller 'fake_moveit_ee_controller' with joints [ ]  
[ INFO] [1505957647.807434551]: Returned 3 controllers in list  
[ INFO] [1505956629.889136115]: Loading robot model 'ur5'...  
[ WARN] [1505956629.889226914]: Skipping virtual joint 'virtual_joint' because its child frame 'box_fixed_link' does not match the URDF frame 'world'  
[ INFO] [1505956631.143209340]: No active joints or end effectors found for group ''. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.  
[ INFO] [1505956631.143508371]: No active joints or end effectors found for group 'gripper'. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.  
[ INFO] [1505956631.212833955]: No active joints or end effectors found for group 'gripper'. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.  
[ INFO] [1505956631.213958002]: Constructing new MoveGroup connection for group 'gripper' in namespace ''  
[ INFO] [1505956633.034567194]: Ready to take commands for planning group gripper.  
```
It seems that I got a group named '', it's so strange!

I make a  robot using a fixed box as the human-like truck, a UR5 robot arm and a Kinect as human-like eyes.

My URDF model please see [here]()

[image]

My SRDF file is:
```
  <?xml version="1.0" ?>
  <!--This does not replace URDF, and is not an extension of URDF.
  This is a format for representing semantic information about the robot structure.
  A URDF file must exist for this robot as well, where the joints and the links that are referenced are defined
  -->
  <robot name="ur5">
  <!--GROUPS: Representation of a set of joints and links. This can be useful for specifying DOF to plan for, defining arms, end effectors, etc-->
  <!--LINKS: When a link is specified, the parent joint of that link (if it exists) is automatically included-->
  <!--JOINTS: When a joint is specified, the child link of that joint (which will always exist) is automatically included-->
  <!--CHAINS: When a chain is specified, all the links along the chain (including endpoints) are included in the group. Additionally, all the joints that are parents to included links are also included. This means that joints along the chain and the parent joint of the base link are included in the group-->
  <!--SUBGROUPS: Groups can also be formed by referencing to already defined group names-->
  <group name="ur5manipulator">
      <chain base_link="base_link" tip_link="ee_link" />
  </group>
  <group name="gripper">
      <link name="bh_base_link" />
      <link name="bh_finger_11_link" />
      <link name="bh_finger_12_link" />
      <link name="bh_finger_13_link" />
      <link name="bh_finger_21_link" />
      <link name="bh_finger_22_link" />
      <link name="bh_finger_23_link" />
      <link name="bh_finger_31_link" />
      <link name="bh_finger_32_link" />
      <link name="bh_finger_33_link" />
  </group>
  <group name="moveit_ee">
      <link name="ee_link" />
  </group>
  <!--GROUP STATES: Purpose: Define a named state for a particular group, in terms of joint values. This is useful to define states like 'folded arms'-->
  <group_state name="InitPose" group="ur5manipulator">
      <joint name="elbow_joint" value="0" />
      <joint name="shoulder_lift_joint" value="0" />
      <joint name="shoulder_pan_joint" value="0" />
      <joint name="wrist_1_joint" value="0" />
      <joint name="wrist_2_joint" value="0" />
      <joint name="wrist_3_joint" value="0" />
  </group_state>
  <!--END EFFECTOR: Purpose: Represent information about an end effector.-->
  <end_effector name="moveit_eef" parent_link="ee_link" group="gripper" parent_group="moveit_ee" />
[... 78 lines cut ...]
  <disable_collisions link1="box_fixed_link" link2="kinect_link" reason="Adjacent" />
  <disable_collisions link1="box_fixed_link" link2="shoulder_link" reason="Never" />
  <disable_collisions link1="ee_link" link2="wrist_1_link" reason="Never" />
  <disable_collisions link1="ee_link" link2="wrist_2_link" reason="Never" />
  <disable_collisions link1="ee_link" link2="wrist_3_link" reason="Adjacent" />
  <disable_collisions link1="forearm_link" link2="upper_arm_link" reason="Adjacent" />
  <disable_collisions link1="forearm_link" link2="wrist_1_link" reason="Adjacent" />
  <disable_collisions link1="kinect_link" link2="shoulder_link" reason="Never" />
  <disable_collisions link1="shoulder_link" link2="upper_arm_link" reason="Adjacent" />
  <disable_collisions link1="wrist_1_link" link2="wrist_2_link" reason="Adjacent" />
  <disable_collisions link1="wrist_1_link" link2="wrist_3_link" reason="Never" />
  <disable_collisions link1="wrist_2_link" link2="wrist_3_link" reason="Adjacent" />
  </robot>
```

And I can't find a way to debug it.

My system configuration:  
```
Ubuntu: 16.04  
ROS: kinetic  
Moveit: 0.9.9-Alpha
```

Any help will be appreciated!

v4hn, 2017-09-21:
>   <end_effector name="moveit_eef" parent_link="ee_link" group="gripper" parent_group="moveit_ee" />

Having a group `moveit_ee` in between that does nothing really seems weird here, especially because the specified link already is part of the `ur5manipulator` group.
Do you get better results if you set that one as `parent_group` here?

I have no idea where the `''` group comes from at the moment.

General stabbing into the dark:

Did you choose `ur5manipulator` as your planning group in rviz? Looks like it could be `gripper`, which doesn't specify end-effectors.

Has the machine you run rviz on the inverse kinematics plugin installed that you configured in `kinematics_base.yaml`?

jacknlliu, 2017-09-22:
@v4hn  Thank you very much for your reply.

> Do you get better results if you set that one as `parent_group` here?

I follow your idea, and remove `moveit_ee` which does nothing then use `gripper` as end-effector.

But I still get the following logs.
```
[ INFO] [1506047757.421200147]: No active joints or end effectors found for group ''. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1506047757.421318406]: No active joints or end effectors found for group 'gripper'. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1506047757.460379173]: No active joints or end effectors found for group 'gripper'. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1506047757.463120623]: Constructing new MoveGroup connection for group 'gripper' in namespace ''
```

> Did you choose `ur5manipulator` as your planning group in `rviz`? Looks like it could be `gripper`, which doesn't specify `end-effectors`.

Yes, the default plan request robot is `gripper`, so I cannot get the interactive markers. After I switch to `ur5manipulator`, I should change the interactive marker size from 0 to 0.5 manually, is this the default behavior?

> Has the machine you run rviz on the inverse kinematics plugin installed that you configured in `kinematics_base.yaml`?

I only have `kinematics.yaml`, and I choose `kdl_kinematics_plugin/KDLKinematicsPlugin` as kinematics solver, and how to choose a ubuntu package for this configuration?

My `SRDF` file is:
```
<?xml version="1.0" ?>
<!--This does not replace URDF, and is not an extension of URDF.
    This is a format for representing semantic information about the robot structure.
    A URDF file must exist for this robot as well, where the joints and the links that are referenced are defined
-->
<robot name="ur5">
    <!--GROUPS: Representation of a set of joints and links. This can be useful for specifying DOF to plan for, defining arms, end effectors, etc-->
    <!--LINKS: When a link is specified, the parent joint of that link (if it exists) is automatically included-->
    <!--JOINTS: When a joint is specified, the child link of that joint (which will always exist) is automatically included-->
    <!--CHAINS: When a chain is specified, all the links along the chain (including endpoints) are included in the group. Additionally, all the joints that are parents to included links are also included. This means that joints along the chain and the parent joint of the base link are included in the group-->
    <!--SUBGROUPS: Groups can also be formed by referencing to already defined group names-->
    <group name="ur5manipulator">
        <chain base_link="base_link" tip_link="ee_link" />
    </group>
    <group name="gripper">
        <link name="bh_base_link" />
        <link name="bh_finger_11_link" />
        <link name="bh_finger_12_link" />
        <link name="bh_finger_13_link" />
        <link name="bh_finger_21_link" />
        <link name="bh_finger_22_link" />
        <link name="bh_finger_23_link" />
        <link name="bh_finger_31_link" />
        <link name="bh_finger_32_link" />
        <link name="bh_finger_33_link" />
    </group>
    <!--GROUP STATES: Purpose: Define a named state for a particular group, in terms of joint values. This is useful to define states like 'folded arms'-->
    <group_state name="InitPose" group="ur5manipulator">
        <joint name="elbow_joint" value="0" />
        <joint name="shoulder_lift_joint" value="0" />
        <joint name="shoulder_pan_joint" value="0" />
        <joint name="wrist_1_joint" value="0" />
        <joint name="wrist_2_joint" value="0" />
        <joint name="wrist_3_joint" value="0" />
    </group_state>
    <!--END EFFECTOR: Purpose: Represent information about an end effector.-->
    <end_effector name="moveit_eef" parent_link="ee_link" group="gripper" parent_group="ur5manipulator" />
    <!--VIRTUAL JOINT: Purpose: this element defines a virtual joint between a robot link and an external frame of reference (considered fixed with respect to the robot)-->
    <virtual_joint name="virtual_joint" type="fixed" parent_frame="world" child_link="box_fixed_link" />
    <!--DISABLE COLLISIONS: By default it is assumed that any link of the robot could potentially come into collision with any other link in the robot. This tag disables collision checking between a specified pair of links. -->
    <disable_collisions link1="base_link" link2="box_fixed_link" reason="Adjacent" />
    <disable_collisions link1="base_link" link2="kinect_link" reason="Never" />
    <disable_collisions link1="base_link" link2="shoulder_link" reason="Adjacent" />
    <disable_collisions link1="bh_base_link" link2="bh_finger_11_link" reason="Adjacent" />
    <disable_collisions link1="bh_base_link" link2="bh_finger_12_link" reason="Never" />
    <disable_collisions link1="bh_base_link" link2="bh_finger_21_link" reason="Adjacent" />
[... 58 lines cut ...]
    <disable_collisions link1="bh_finger_31_link" link2="wrist_2_link" reason="Never" />
    <disable_collisions link1="bh_finger_31_link" link2="wrist_3_link" reason="Never" />
    <disable_collisions link1="bh_finger_32_link" link2="bh_finger_33_link" reason="Adjacent" />
    <disable_collisions link1="bh_finger_32_link" link2="ee_link" reason="Never" />
[... cut ...]
    <disable_collisions link1="forearm_link" link2="wrist_1_link" reason="Adjacent" />
    <disable_collisions link1="kinect_link" link2="shoulder_link" reason="Never" />
    <disable_collisions link1="shoulder_link" link2="upper_arm_link" reason="Adjacent" />
    <disable_collisions link1="wrist_1_link" link2="wrist_2_link" reason="Adjacent" />
    <disable_collisions link1="wrist_1_link" link2="wrist_3_link" reason="Never" />
    <disable_collisions link1="wrist_2_link" link2="wrist_3_link" reason="Adjacent" />
</robot>
```

ysl208, 2017-11-02:
I'm having the same issue and no idea how to fix it. id you manage to solve it?
I still get the error
`[ INFO] [1505956631.143209340]: No active joints or end effectors found for group ''. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.  `

jacknlliu, 2017-11-03:
I think a key point is that how or where moveit get the group namespace.

ysl208, 2017-11-03:
I cleaned my whole workspace and set it up from scratch, now the error doesn't occur anymore.

saurabhbansal90, 2017-12-27:
I have cleared and started from scratch, but same error is coming for me.
Everytime it through this end-effector error.

Please let me know  if somebody has a solution for this.

zhuyiif, 2017-12-30:
+1 some problem, I don't know how to fix it

philwall3, 2018-01-04:
I also have that problem.

Heisenberg1997, 2018-01-18:
wtf ,  the problem is still not being fixed?

saurabhbansal90, 2018-02-14:
There should be some problem for this problem. Please provide guidance to solve this.

rhaschke, 2018-02-14:
If you want help on this problem, you should provide more details. Please provide your URDF and your complete moveit_config package.

saurabhbansal90, 2018-02-15:
Hello rhaschke, Thanks for the response. Please find my moveit_config package at this link[https://drive.google.com/open?id=1Qfrf703LU4jHA-Y0AmwcWsmKvIXJ1FmG](url).
And my urdf at this link[https://drive.google.com/file/d/1B1GIPtXs5Pwbd6lS_LECPnTiLVXDD5EU/view?usp=sharing](url)

Please let me know if any other information  is needed.

rhaschke, 2018-02-15:
@saurabhbansal90 Thanks for providing the files. I'm afraid that I also need the complete robot_description_pkg including the meshes to see something. However, I can already reproduce the error and will have a look.

rhaschke, 2018-02-15:
The empty group name is only a cosmetic artifact and not really an issue.
@saurabhbansal90 You essentially missed to define a kinematics solver for your Arm group, which is why you don't see an interactive marker to move the robot's eef. I will improve warnings correspondingly.

rhaschke, 2018-02-15:
Fixes in #769.

closed 2018-02-15 by rhaschke

saurabhbansal90, 2018-04-01:
Is it always necessary to  use some kinematic solver for  MoveIt package. Can we work without that also.
I was not able to use the MoveIt code because of the errors we were getting as stated, so we decided to use the topics for moving the joints.

But now I got sometime to better understand the problem, so I created a simple project in ROS Kinetic with only a single joint at this [link](https://github.com/saurabhbansal90/my_robot.git).

In this project also, I have not used any Kinematic solver, I tried it also, but same problem.
And as I was having only one joint, I didn't had an end-effector. So for just name sake, I made the child link as my end-effector and it's parent group as the same as MoveGroup 'Arm'.

I am getting logs as below:
`[ INFO] [1522564606.612109259]: Loading robot model 'my_robot'...
[ WARN] [1522564606.612395934]: Could not identify parent group for end-effector 'eef'
[ INFO] [1522564608.010117654, 814.066000000]: Ready to take commands for planning group Arm.
[ INFO] [1522564608.247749097, 814.300000000]: Current Pose: 0.000000, 0.000000, 0.062500
[ INFO] [1522564608.247812996, 814.300000000]: Replanning: yes
[ INFO] [1522564608.247837388, 814.300000000]: Planning for the group
[ WARN] [1522564608.248846338, 814.300000000]: Fail: ABORTED: Catastrophic failure
[ INFO] [1522564613.249187265, 819.180000000]: Executing the plan
[ INFO] [1522564613.249855460, 819.181000000]: ABORTED: Solution found but controller failed during execution
[ INFO] [1522564613.249888230, 819.181000000]: Moving as per Plan
[ INFO] [1522564613.272832947, 819.201000000]: ABORTED: Catastrophic failure`

And below messages I am getting in the window, in which I started MoveIt and RViz:
`[ERROR] [1522563897.263050199, 111.301000000]: Unable to construct goal representation
[ INFO] [1522564608.248200062, 814.300000000]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ERROR] [1522564608.248450997, 814.300000000]: Unable to construct goal representation
[ INFO] [1522564613.249555896, 819.181000000]: Execution request received for ExecuteTrajectory action.
[ WARN] [1522564613.249585377, 819.181000000]: The trajectory to execute is empty
[ INFO] [1522564613.272038265, 819.200000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1522564613.272161488, 819.200000000]: Planning attempt 1 of at most 5
[ERROR] [1522564613.272361407, 819.200000000]: Unable to construct goal representation`

Can you please have a look on the code, and let us know what I am missing in my code.
Thanks in advance.

saurabhbansal90, 2018-04-07:
Hello Micheal, Thanks for response. But I really need to understand why my package is not able to find the end-effector. I am struggling to solve this.
I tried to make it a simple problem by having only a single joint and didn't used  any kinematic solver. I configured the end-effector as well.
But as you can see, it is not able to find the end-effector. Can you please point me in right direction to correct any mistake in my package [link](https://github.com/saurabhbansal90/my_robot.git).

Regards,
Saurabh

sritee, 2019-12-02:
Any updates @saurabhbansal90 ?

rcywongaa, 2022-04-27:
For my future reference, note that the end effector will not be correctly detected if the kinematic solver for the corresponding group failed to load.
Check your `kinematics.yaml` to make sure you've specified the correct group name and kinematic solver AND check log outputs to make sure there were no errors when loading the kinematic solver
If you see the following line
```
[kinematics_plugin_loader]: The kinematics plugin (your gorup name) failed to load.
```
you are very likely to get the `No active joints or end effectors found for group` error (which also directly leads to the missing interactive marker https://github.com/ros-planning/moveit/issues/1399)