moveit/moveit #622: Feed being spamed with collision function warnings

TroyCordie, 2017-09-27:
### Description

I recently installed ompl stand alone that caused some grief with building moveit so I'm not certan this is a bug or caused by the ompl install. I have since uninstalled ompl and am now getting the following warning spamming my feed.

'Warning: collision function between node type 19 and node type 9 is not supported
Warning: collision function between node type 19 and node type 5 is not supported'

It doesn't appear to affect the functionality but is frustrating as it is had to see what else is going on. The warning appears to be coming from FCL but beyond that, I'm not sure what to do.

Thanks in advance for any help 

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04

v4hn, 2017-09-27:
Sorry, I have never seen this error before.
Maybe another maintainer did?

TroyCordie, 2017-10-03:
I have reinstalled FCL and that has resolved the issue. Sorry, I cant give better advice I'm still not certain what the problem was though I assume it was a version Issue.

closed 2017-10-03 by TroyCordie