moveit/moveit #624: Could not find font Arial in MovableText::setFontName

davetcoleman, 2017-09-28:
### Description

The latest version of Rviz has removed the Arial font and according to https://github.com/ros-visualization/rviz/issues/1152 this needs to be reflected in MoveIt! in order to avoid a segfault:

> https://github.com/ros-visualization/rviz/issues/1152

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04

gavanderhoorn, 2017-09-28:
Related ROS Answers question: [Error when trying to launch moveit created robot model](https://answers.ros.org/question/271750).

wjwwood, 2017-09-28:
> in order to avoid a segfault

It isn't a segfault, it is an unhandled instance of `Ogre::Exception`.

I'll make an announcement on discourse, sorry for the disruption this caused.

v4hn, 2017-09-28:
@davetcoleman so where's the corresponding pull-request that does search&replace?
Sounds easy enough and Liberation is the better font either way :-)
I don't have a 16.04 so I couldn't test it at the moment.

wjwwood, 2017-09-28:
Oh, so I was incomplete in my assumptions, I figure Moveit was only using Arial with Ogre (which is the case for rviz), but it's actually using it with Qt too, e.g.:

https://github.com/ros-planning/moveit/blob/4bc43c56ca2a5293f3ef8d55796ec586cfde5f69/moveit_setup_assistant/src/widgets/header_widget.cpp#L60

So in that case Moveit may need to depend on the liberation-sans font package in ubuntu or something. I'm not sure if it will automatically use the one distributed by rviz since it is only registered with Ogre. However, it might "just work" if Ogre is registering the font with freetype or something under the hood (making it implicitly available for Qt as well).

wjwwood, 2017-09-28:
> However, it might "just work" if Ogre is registering the font with freetype or something under the hood (making it implicitly available for Qt as well).

Thinking on it more, that must be the case since Qt wouldn't have found Arial in the past otherwise.

wjwwood, 2017-09-28:
If you choose to just depend on it, then maybe a new rosdep key for https://packages.ubuntu.com/xenial/fonts-liberation would fix the issue (along with find-replace `Arial` with `Liberation Sans` or something similar).

v4hn, 2017-09-28:
> If you choose to just depend on it, then maybe a new rosdep key for https://packages.ubuntu.com/xenial/fonts-liberation

I would be in favor of that.
Could we also depend on another font that has no copyright issues and is distributed in the core of most linux distributions?
To integrate nicely with rviz I guess it makes sense to use Liberation.
Why didn't you make rviz depend on the package in the first place instead of bundling it with the package?

wjwwood, 2017-09-28:
> Why didn't you make rviz depend on the package in the first place instead of bundling it with the package?

Mostly consistency and portability. I'm working with rviz on mac and Windows these days and bringing the font with me is easier to ensure than relying on external dependencies.

Also, I already had code which loaded it from the package (as Arial) and so the minimal change was to distribute Liberation Sans and load it in the same way.

mukesh8101997, 2017-10-12:
can anyone please explain how to change arial font to Liberation san.
It will be very helpful for me.

v4hn, 2017-10-12:
William explained what has to be done in https://github.com/ros-planning/moveit/issues/624#issuecomment-332929567 .

I'll flag this for WMD. Anyone interested in this should have seen `rosdep` before I guess.

mukesh8101997, 2017-10-12:
I have already changed arial into Liberation san in my ogre_media but still getting this error. I don't know how you are using rosdep to remove this issue.
please help a bit more.
I am using ros kinetic though.

v4hn, 2017-10-12:
rosdep helps to solve this problem in a generic way in the packaging system. It doesn't help you with your problem.

I believe 1. you should install the liberation-fonts package of your distribution and 2. change all the strings that say "Arial" to "Liberation Sans". I haven't tested it myself yet.
This is what William said above.

fatihirim, 2017-10-16:
Can you be more specific on what you mean by replacing all Arial with Liberation Sans? Where exactly should I do the replacement? Which file(s)?

Thanks in advance.

v4hn, 2017-10-17:
I guess `git grep Arial` should list all places.

mukesh8101997, 2017-10-17:
@fatihirim 
if you find any solution then please guide others. I am also struggling with this problem for many days.
Thanks.

fatihirim, 2017-10-17:
Unfortunately no luck yet. I have been struggling for two days now. What is strange for me is that I can run Rviz for one robot model created in MoveIt, but cannot run another very similar robot model due to Ogre Font exception regarding Liberation Sans. 

I am using ROS Kinetic on Ubuntu 16.04.

People with more experience on this issue, please provide guidance, or please fix the issue in MoveIt/RViz...

wjwwood, 2017-10-17:
> due to Ogre Font exception regarding Liberation Sans.

Can you please include the exact error? That doesn't sound right.

fatihirim, 2017-10-17:
Here is the exact error:

 what():  OGRE EXCEPTION(5:): Could not find font Liberation Sans in MovableText::setFontName
[rviz_ML_Ubuntu_11752_4341152296783752344-4] process has died [pid 11776, exit code -6, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/ddomit/catkin_ws/src/RoboND-Kinematics-Project/kr210_claw_moveit/launch/moveit.rviz __name:=rviz_ML_Ubuntu_11752_4341152296783752344 __log:=/home/ddomit/.ros/log/c7f078f8-a4de-11e7-95b3-4ccc6acfcea6/rviz_ML_Ubuntu_11752_4341152296783752344-4.log].
log file: /home/ddomit/.ros/log/c7f078f8-a4de-11e7-95b3-4ccc6acfcea6/rviz_ML_Ubuntu_11752_4341152296783752344-4*.log

wjwwood, 2017-10-17:
@fatihirim that's a different issue then, please open a ticket on the rviz repository and include how you have rviz installed, what version it is, os name and version, and if possible simple steps to reproduce the issue.

davetcoleman, 2017-10-18:
I apologize for not submitting a patch for this yet but fyi my dirty work around was to just build rviz locally using the commit before the font was removed from rviz. This is a terrible workaround.

fatihirim, 2017-10-18:
Hi @davetcoleman,
Thanks for this info.
When are you planning to submit the patch? Anytime soon?
I assume you will notify us on this thread when you are finished with the patch.

Thanks in advance!  Eagerly waiting for the patch :)

v4hn, 2017-10-18:
Ok, so I've just seen this bug for the first time myself.
As William said above, there are two different problems and this issue got hijacked by the second group.

The first one (in this issue) is missing **Arial** that is used in `moveit_setup_assistant`.
This has to be fixed in the package.

The second is about missing **Liberation Sans**.
Apparently, the cartographer project experienced a similar bug recently.
It turns out (as previously said to me, but I didn't believe it) that **recompiling the display** solves that problem.
At the moment, I have no idea why....
So this problem can be solved by compiling moveit from source.

davetcoleman, 2017-10-18:
@wjwwood also plans to work on this today

wjwwood, 2017-10-18:
@v4hn can you link to an issue about the cartographer related bug? I don't think there is anything in the rviz headers or cmake that would "bake" in the use of Arial or make Liberation Sans unavailable.

v4hn, 2017-10-18:
I spoke to one of the people (Jihon) involved there in person today.
I'm sorry, but I don't know where or whether this is documented.

jspricke, 2017-10-18:
@jihoonl can you comment?

wjwwood, 2017-10-18:
So, after debugging this with @davetcoleman on a machine that had the problem for him I think both of these errors:

- `OGRE EXCEPTION(5:): Could not find font Liberation Sans in MovableText::setFontName`: reported here https://answers.ros.org/question/271750/error-when-trying-to-launch-moveit-created-robot-model/
- `OGRE EXCEPTION(5:): Could not find font Arial in MovableText::setFontName `: reported here https://github.com/ros-visualization/rviz/issues/1152#issue-260830581

Are in the same class, basically they occur when you load `librviz.so` from one install prefix but `rospack find rviz` returns a different installation prefix _and_ those install prefixes have different versions of rviz across the `1.12.12` version line.

For instance, if you got the `Could not find font Liberation Sans` message, then you have a `librviz.so` which comes from `1.12.12`+ (because the code is looking for liberation sans) and if you run `rospack find rviz` would return a `share` folder which contains `arial.fontdef` rather than `liberation_sans.fontdef`. Therefore you have ogre knowing where arial is, but not liberation sans, and rviz code asking for liberation sans, thus the error.

The inverse can also happen, where the code in `librviz.so` is looking for Arial, but only the `liberation_sans.fontdef` file is loaded by ogre, producing the `Could not find font Arial` message.

In both cases the solution is to make sure your `ROS_PACKAGE_PATH` and `LD_LIBRARY_PATH` agree on where rviz/`librviz.so` are.

----

Separately, the moveit setup assistant (possibly other things) reference Arial directly. After testing it, this doesn't seem to break anything (on my Xenial machine with updated moveit and rviz debs). My guess is that Arial is somehow available through Qt or it automatically provides an alternative.

I'll open a pr to use the system default font instead since it doesn't seem to matter what font we use (@davetcoleman supported this move too).

----

To recap, I think most of the errors are due to inconsistent state on machines and the resolution is to make sure you have up-to-date source and debs and you rebuild your code against the new versions. Also we can remove the use of Arial in moveit, but that doesn't seem to be strictly required.

I tested a few different scenarios and didn't run into any issues with up-to-date debs on Xenial and Kinetic. So, if anyone still has issues, please try to include as much detail as possible about your setup, ideally with reproducible steps and I'll try to investigate it on a step by step basis.

davetcoleman, 2017-10-19:
Thanks @wjwwood - glad we could work on this today.

For the simple answer for anyone facing this problem, make sure ``apt-get dist-update`` is run and you rebuild your whole catkin workspace from scratch (``catkin clean``). This fixed it for me - without having to install Rviz from source.

jihoonl, 2017-10-19:
I also had to deal with `OGRE EXCEPTION(5:): Could not find font Arial in MovableText::setFontName` error while using [cartographer_rviz](https://github.com/googlecartographer/cartographer_ros/tree/master/cartographer_rviz) even if it uses MovableText with a default font from rviz([here](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_rviz/cartographer_rviz/drawable_submap.cc#L64-L85)).

The issue was gone after clean recompilation like @davetcoleman commented(`catkin clean && catkin build`). So, IMHO a static link to find fonts created at compile time are stored in the plugin `.so` file and used at runtime. If it is true, the moveit plugin inevitably fails to look up a font when rviz version is upgraded to (>= 1.12.12) because the moveit debian(0.9.9) that is already present in the system was compilied using old version of rviz(1.12.11) where font was still located in `ogre_media/fonts`.

wjwwood, 2017-10-19:
> So, IMHO a static link to find fonts created at compile time are stored in the plugin .so file and used at runtime.

That's part true, which font to look up, `Arial` or `Liberation Sans`, is embedded in the `librviz.so` shared object (not statically linked but multiple may exist on the system with different source versions). The location of the fonts is looked up dynamically at run time using the `rospack` API which looks up the location of packages based on the `ROS_PACKAGE_PATH`.

wjwwood, 2017-10-19:
Here's the pr to replace explicit uses of Arial from the setup assistant, though as I said before I don't think this fixes any problems, it's just a good idea: https://github.com/ros-planning/moveit/pull/668

mat-l, 2017-10-24:
Dear all,

I executed the following command on Ubuntu 14.04 ROS Indigo, but I still cannot start Rviz :(
apt-get dist-upgrade
apt-get install fonts-liberation
catkin_make clean
catkin_make

Any other ideas how to solve the issue easily or when can we expect updated MoveIt packages?

wjwwood, 2017-10-24:
@mat-l as I stated, there is no required update to the moveit packages due the rviz change. Please follow my recommendation and tell us about your workspace and how you set it up, what you've given us is not enough information to help.

I wrote (https://github.com/ros-planning/moveit/issues/624#issuecomment-337751373):
> To recap, I think most of the errors are due to inconsistent state on machines and the resolution is to make sure you have up-to-date source and debs and you rebuild your code against the new versions. Also we can remove the use of Arial in moveit, but that doesn't seem to be strictly required.

> I tested a few different scenarios and didn't run into any issues with up-to-date debs on Xenial and Kinetic. So, if anyone still has issues, please try to include as much detail as possible about your setup, ideally with reproducible steps and I'll try to investigate it on a step by step basis.

GiuCotugno, 2017-11-11:
For those of you still interested on this, I managed to make to work rviz and moveit! together on ROS indigo as I got a similar problem to above: rviz/moveit could not find liberation sans but it could find arial font. The problem arose using Baxter robot (if it is at all relevant)

The steps I performed are the following:
1) download and install liberation sans fonts (you can find them here https://pagure.io/liberation-fonts)
2) move LiberationSans-Regular.ttf to /opt/ros/indigo/share/rviz/ogre_media/fonts (which is the folder where rviz's fonts are)
3) create a new file and call it "liberationsans.fontdef"
4) copy paste the below text in it:
Liberation Sans
{
  type truetype
  source LiberationSans-Regular.ttf
  size 18
  resolution 96
}

Pay extra attention to the name of the block (Liberation Sans), as rviz/moveit look at that to find the font (apparently)

If you restart rviz and add a movit element into it, should now work (or at least it does for me). It seems that Rviz won't be able to find a font if you don't define that file. Hope this might be of help for you as well