moveit/moveit #625: Could not build moveit from source : no matching function for fcl library

thibaultbarbie, 2017-09-29:
I tried to build moveit from source with Ubuntu 16.04 and ros kinetic version but it seems to have a problem with fcl. 

When I tried to compile it after having installed fcl (http://moveit.ros.org/install/source/dependencies/) it gave me : 

`Finished  <<< libfcl-dev                                       [ 2 minutes and 10.3 seconds ]`  
`Starting  >>> moveit_core`              
`Errors     << moveit_core:make /home/thibault/ros_ws/logs/moveit_core/build.make.000.log`            
`/home/thibault/ros_ws/src/moveit/moveit_core/collision_detection_fcl/src/collision_robot_fcl.cpp: In constructor ‘collision_detection::CollisionRobotFCL::CollisionRobotFCL(const RobotModelConstPtr&, double, double)’:`
`/home/thibault/ros_ws/src/moveit/moveit_core/collision_detection_fcl/src/collision_robot_fcl.cpp:62:102: error: no matching function for call to ‘fcl::CollisionObject::CollisionObject(const std::shared_ptr<fcl::CollisionGeometry>&)’`
         `fcl_objs_[index] = FCLCollisionObjectConstPtr(new fcl::CollisionObject(g->collision_geometry_));`
                                                                                                      `^`
`In file included from /usr/local/include/fcl/broadphase/broadphase.h:43:0,
                 from /home/thibault/ros_ws/src/moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:43,
                 from /home/thibault/ros_ws/src/moveit/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_robot_fcl.h:40,
                 from /home/thibault/ros_ws/src/moveit/moveit_core/collision_detection_fcl``/src/collision_robot_fcl.cpp:37:`
`/usr/local/include/fcl/collision_object.h:171:3: note: candidate: fcl::CollisionObject::CollisionObject(const boost::shared_ptr<fcl::CollisionGeometry>&, const Matrix3f&, const Vec3f&)`

So fcl seems to be compiled but moveit cannot find it.

To install moveit I followed the install tutorial in http://moveit.ros.org/install/source/

knorth55, 2017-10-11:
I had same trouble, and I got the information that moveit requires `0.5.0` of `fcl`.
https://discourse.ros.org/t/fcl-0-5-packages-for-moveit-in-kinetic/364

v4hn, 2017-10-11:
looking into the error messages above shows that you installed an old version of `fcl`, @knorth55 is right.
Closing as invalid.

closed 2017-10-11 by v4hn