moveit/moveit #629: Reduce memory allocations in IK solvers

davetcoleman, 2017-10-11:
It was requested in [pull request](https://github.com/ros-planning/moveit/pull/537#issuecomment-320294696) by @Jmeyer1292  that the IKFast and KDL IK solvers reduce memory allocation during runtime and instead only allocate at initialization. Even though we do not make any runtime guarantees, IK is called so frequently during some types of planning that we would see large speedups.

This changes should be relatively simple and ideal for a World MoveIt! Day project. Be sure to test against [moveit_kinematics_tests](https://github.com/ros-planning/moveit_kinematics_tests/) for any breakages.

v4hn, 2017-10-11:
Great idea!

I believe this is not simple though. Things can get ugly though if you look into thread-safety
- I'm uncertain about whether we currently guarantee it, but we want to aim for it either way.

gavanderhoorn, 2017-10-11:
I would suggest to 'require' that a potential taker of this enhancement first profile a typical kinematics solver to see whether there is a lot of overhead in however many dynamic allocations they might do.

I trust @Jmeyer1292, but it would be a shame if someone spends their WMD figuring out how to avoid all sorts of allocs to in the end not really see any large improvement.

Do you have any nrs @Jmeyer1292 ?

Jmeyer1292, 2017-10-11:
My comment was more specifically targeted at the PR that @davetcoleman mentioned because it seemed to do a lot of work for little gain: it made an extra vector and sorted it when all that needed to happen was to linearly march over existing data and remember the best item.

To the point made by @v4hn, MoveIt's API makes removing allocations difficult in a threaded environment. I wouldn't bother with it.

v4hn, 2017-10-11:
Nevertheless, I believe at least *profiling* the current kinematics plugins to see how much it could improve is a good idea for WMD.

marip8, 2017-10-18:
I will start working on this

marip8, 2017-10-18:
[image]

I created a test node that solves a Motoman ES200RN-120's kinematics at 10k unique poses (in a ~1x1m vertical grid in front of the robot) using the IKFast plugin, and I profiled it with valgrind.

The results show that the `getPositionIK` function spends about 2.5% of its time sorting the list of potential solutions to get the closest solution to the seed. That doesn't seem to be too much of a time-sink, but there does seem to be a lot of time spent in other overhead functions (i.e ~20% of time in `getSolution`, etc.)

Jmeyer1292, 2017-10-19:
Definitely less time than I thought, but still enough to be appreciable.

The real stand out here is the crazy amount of overhead associated with IKFast's API. A `std::list`, really? It seems like maybe 25% of the time is spent doing math.

davetcoleman, 2017-10-19:
Thanks for looking into this!

simonschmeisser, 2017-11-12:
slightly offtopic: can you provide a link/repo for how you created that measurement and that fancy graphic?

marip8, 2017-11-13:
For the profiler I used [valgrind][1], and the graphic was created using [kcachegrind][2]

[1]: http://valgrind.org
[2]: http://kcachegrind.sourceforge.net/html/Home.html