moveit/moveit #631: MoveGroup::Place ignores goal poses specified in end-effector frame.

mbharatheesha, 2017-10-12:
### Description
MoveGroup::Place interprets goal poses in the object reference frame despite setting `place_eef` to True in the `PlaceGoal` msg.

[Reference discussion on moveit-users google group](https://groups.google.com/forum/#!topic/moveit-users/MTLEB9diPMs)

[Related reference](https://groups.google.com/forum/#!searchin/moveit-users/place$20pose/moveit-users/_2_MGQ4jy7A/VDMxOABDY60J)

### Environment
* ROS Distro: [Indigo]
* OS Version: e.g. Ubuntu 14.04

### Steps to reproduce
* Specify a place pose in the end-effector frame and set the place_eef flag to True.
* Call the MoveGroup Place API using group.place()

### Expected behaviour
The place call should succeed and the robot (in simulation or reality) should move to the desired location and perform the desired action (unless there are obvious collisions)

### Actual behaviour
Non-obvious collisions are reported and place call fails.

v4hn, 2017-10-12:
Hey @mbharatheesha , thank you for the report.
I know the structure of the relevant code is not easy, but did you have a look at it?

Do you have an easy-to-setup example for this at hand?

It is definitely possible that this is a mistake in the code, but it might also break for your setup only.

mbharatheesha, 2017-10-12:
@v4hn:
> I know the structure of the relevant code is not easy, but did you have a look at it?

Yes, I did. And it comes from [here](https://github.com/ros-planning/moveit/blob/069f6c83f75c630b651350b678f45a0f7b6ec3d5/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L681).

Basically, the complete goal information from the user defined PlaceGoal msg is not used.

> Do you have an easy-to-setup example for this at hand?

Unfortunately not. But, this is something that can be produced on any setup by using the MoveGroup `place` API instead of the MoveIt place capability. More details are in this [discussion](https://groups.google.com/forum/#!topic/moveit-users/MTLEB9diPMs)

> It is definitely possible that this is a mistake in the code.

I think it is. I did fix it on my source installation, but I am not very proud of the fix because I basically bulldozed the goal construction by setting the `place_eef` flag. I did this because it was in a rush for another project.

But will prepare a clean PR for this hopefully during WMD17.

v4hn, 2017-10-12:
In that case I'll mark this as WMD candidate and assign it to you :-)

Thanks a lot in advance, I'm looking forward to see this cleaned up!

mbharatheesha, 2017-10-18:
I have been working on this. The corresponding changes are available [here](https://github.com/mbharatheesha/moveit/commit/d01a91c4566ddb89b2b824cf8cd7179ba1bd7a57).

However, I have hit the following roadblock:
The call to place via MoveGroupInterface fails with [this issue] (https://github.com/ros-planning/moveit_tutorials/issues/112). I do get the appropriate group name with group.getName(). So, I am not sure what is happening here.

I have been investigating for a couple of hours now why this issue pops up. But no progress yet.

Any thoughts?

After updating to the latest upstream I would still like to do the following:

- create a corresponding python API as well.
- update documentation on the current place API.
- Test the changes after the above mentioned issue is fixed.

So, I guess, this won't be a PR for today :-|

v4hn, 2017-10-18:
fair enough, I noticed in advance that the corresponding code flow is rather hard to follow...
Thank you for your support here! Of course this problem will remain on the radar (and hopefully be solved soon...)

mvieth, 2019-05-30:
Has anyone looked into this again? I think this is worth revisiting. The issue @mbharatheesha ran into has been closed in the meantime, and was unrelated I believe. The changes @mbharatheesha proposed look good in my opinion, maybe they are ready for a PR? Anything else I can do?