moveit/moveit #634: MoveGroupInterface copy-constructor

v4hn, 2017-10-13:
The `MoveGroupInterface` wraps an implementation class, but does not consider this for copy-construction. The class does not define a copy-constructor, thus the default one is used and copies the `impl_` *pointer*, but the destructor just deletes the referenced object nevertheless.

This creates the following bug that comes up every now and then with students:
```
void f(moveit::planning_interface::MoveGroupInterface mgi){
   return;
}

int main(int argc, char** argv){
   ros::init(argc, argv, "test");

   moveit::planning_interface::MoveGroupInterface mgi("arm");
   f(mgi);
   // mgi->impl_ has been freed => future calls to mgi methods will break/segfault
```
This can be worked around by changing the parameter of `f` into a reference - which is the more proper way to do it in this case anyway.

Nevertheless I believe we want to have a working copy-constructor here.

Jmeyer1292, 2017-10-18:
I'll take a look at this one to start off with.

Jmeyer1292, 2017-10-18:
@v4hn Do you think it should be copy constructible? We have a few options:
1. Disable the copy constructor. Safe, minimal effort, preserves correctness. Could break code thats already invalid.
2. Allow the copy constructor to share a reference counted pointer. Can produce bugs through aliasing.
3. Copy the underlying implementation class. This itself has lots of member variables, many of which may not be copyable themselves.

I'll start down option 3, but consider option 1: it may be the more correct thing to do with a class like this.

Jmeyer1292, 2017-10-18:
To illustrate the complications, the move group implementation manages things like a thread for constraint sampling, ~4 action clients (which are not copyable), db handles, etc

jrgnicho, 2017-10-18:
Going for your 3rd option seems reasonable to me.

v4hn, 2017-10-18:
I prefer 3., but 1. is a "cheap" solution if you encounter too many problems.

davetcoleman, 2017-10-18:
I think I prefer option 1 because some variables are not copyable in option 3

closed 2019-09-06 by v4hn