moveit/moveit #636: optionally wait for controllers indefinitely

v4hn, 2017-10-13:
During startup, MoveIt's `move_group` node waits for the external controllers (`FollowJointTrajectory` or `GripperCommand` action servers) that have been specified in `config/controllers.yaml`.

However, if they do not come up within some fixed time, they are ignored, `move_group` continues, and they will not be available later on. MoveIt will be unable to execute trajectories for the relevant joints.

I would propose to make this behavior - for backward compatibility - optional via a ros-parameter.

See [here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/action_based_controller_handle.h#L76) for the code that handles this.

ipa-bfb, 2017-10-18:
Me and @ipa-aep can take care of this...

v4hn, 2017-11-24:
Merged in #695 .Thanks a lot for the contribution!

closed 2017-11-24 by v4hn