moveit/moveit #639: setFromIK with Collision Checking

v4hn, 2017-10-13:
[Our tutorial on setFromIK](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/kinematics/src/doc/kinematic_model_tutorial.html#inverse-kinematics) does not use collision checking.

How to add it in there has been a common question in the last few years.
An involved example that does it can be found [here](https://github.com/TAMS-Group/moveit_task_constructor/blob/master/src/subtasks/generate_grasp_pose.cpp#L175).
As you can see there, it is not very straight-forward and requires more infrastructure.

I propose to add a new method overload `isStateValid` to the `PlanningScene` class that can be directly passed in as the `constraint` parameter for `setFromIK`.
This should be tested (obviously) and documented in the linked tutorial.

MoveIt separates Collision Checking and Kinematics capabilities. The earlier are part of the `PlanningScene` class, the latter part of `RobotState` and `RobotModel`.
All of these classes can be found in `moveit_core`.

davetcoleman, 2017-10-18:
An easier example for using setFromIK with collision checking can be found in this fairly [simple interactive marker](https://github.com/ros-planning/moveit_visual_tools/blob/kinetic-devel/include/moveit_visual_tools/imarker_robot_state.h) for dual arm IK demo. This would be good to add to the tutorials, also.

jrgnicho, 2017-10-18:
Would just a better example of how to do collision checking be sufficient?

v4hn, 2017-10-18:
I believe we should have a tutorial that enables users to do collision-aware ik, not just do collision checking.

jschleicher, 2018-12-03:
The first link to moveit_task_constructor is dead, permalink probably mentions the line to use [isValid callback in setFromIK](https://github.com/ros-planning/moveit_task_constructor/blob/1b1a82f7c8634d3b7f704d8ec22e624ef4c3c431/src/subtasks/generate_grasp_pose.cpp#L168).

+1 for adding a simple to use overload together with documentation.

davetcoleman, 2018-12-03:
Anyone is welcome to contribute to the tutorials! its not very difficult ;-)

buckleytoby, 2023-07-27:
has there been any updates on this? Even in moveit2 (ROS2 Humble) I don't think setFromIK is obstacle-aware. You can pass in a GroupStateValidityCallbackFn pointer however, and that could point to moveit's own collision checker?