moveit/moveit #640: Passive Joints not Excluded from Joint Trajectory

rkeatin3, 2017-10-13:
### Description

Passive joints included in a kinematic chain are not excluded in the generated joint trajectory when planning motion. 

### My environment
* ROS Distro: Jade
* OS Version: Ubuntu 14.04
* Binary Package Version 0.8.7

### Steps to reproduce
I have attached a package that shows the behavior. You can see that in fake_controllers.yaml the compliant joint is included. If taken out, attempting to move the robot via the demo launch file will result in failure with the typical `Unable to identify any set of controllers that can actuate the specified joints` message.
[moveit_issue.zip](https://github.com/ros-planning/moveit/files/1384157/moveit_issue.zip)

### Expected behaviour
Ability to plan trajectories over chains with passive joints. To be honest, though, I'm not sure how I expect the collision checking to work with passive joints. 

### Actual behaviour
Inability to plan trajectories over chains with passive joints.

rfn123, 2017-10-26:
Had the same problem, I ended up ignoring the commands passed to the passive joint in the hardware interface since I was using position control. Would be nice though if such scenario could be handled.

v4hn, 2017-11-24:
Sorry for the late response.

I am unsure, conceptually, whether this is correct or broken behavior.

If your passive joint is actually coupled and controlled through the active joints in the group, e.g. in the Robotiq 3-Finger Adaptive Gripper, it makes some sense to forward the "planned trajectory" for it.
The controller should ignore it if it doesn't now how to make use of the information.

If, however, the passive joint is independent, it does not belong in the joint trajectory for the group, I agree.

Comments or strong opinions anyone?

bmagyar, 2017-11-24:
This reminds me to set up a unit test that makes sure Moveit is compatible with the latest version of `joint_trajectory_controller` from `ros_control`. 

I think passive joints is a slightly vague term here. They could be fixed, mimic or defined by some complex rule one prefers to ignore.

I agree with forwarding the planned passive joint trajectory but only for visualisation, the controllers should only be given commands to joints that can be controlled.

rkeatin3, 2017-11-24:
So I can explain why I created this issue if it helps determine the solution: I am controlling a vacuum gripper that has a compliant prismatic joint to ensure good contact with the object to be picked up. I have it mounted on a UR5, and ideally, I want to kinematic chain for my manipulator to go from its base link all the way to the coordinate frame at the end of the gripper (for ease of use in general but also due to [Issue 577](https://github.com/ros-planning/moveit/issues/577)).

With that setup, plans sent out to the controller for UR5 have the passive joint included, and I don't particularly want to fork and modify that. Using the joints interface to specify the move group could be a workaround, but I have been using KDL and Trac-IK (which require kinematic chains).

bmagyar, 2017-11-25:
Thanks for the clarification Ryan.

Do you need moveit to plan with the joints of the gripper?
I used a similar setup for years with heavily underactuated hands.

What I used was to only set up the kinematic chain until an arbitrary
grasping frame around the palm (vaguely in the centre of the hand
workspace) and leave the gripper/hand joints put of the planning group. The
end effector will still be part of the plan as per collision avoidance but
won't play a role in the actuation. Hope I didn't misunderstand your case
:)

<https://github.com/ros-planning/moveit/issues/577>).

rkeatin3, 2017-11-27:
I believe I see what you're saying. Define a grasp link, so to speak, with respect to the end of the manipulator that is coincident to link I care to plan with? Something like: 

          Manipulator EE Link
               /                  \
Passive Joint              |
             |                     |
Gripper EE Link      Grasp Link

That should work to side-step this issue with passive joints being passed in the trajectory.

bmagyar, 2017-11-28:
It doesn't only sidestep the passive joint issue but also avoids planning with joints you don't care about be it passive or active joints. I do the same with parallel grippers.

v4hn, 2017-11-28:
I believe this is not what Bence said, but it could be a dirty workaround, yes.

It is rather easy though (basically an "if" in the right place) to ignore passive joints that are not part of any controller.
Given the discussion here it sounds like a great addition (and I expected it to behave that way).

rkeatin3, 2017-12-08:
@v4hn, where are you suggesting that this if statement go? If your intent was to put it in MoveIt (before it sends to plan to the controller), then I agree.

v4hn, 2017-12-10:
> where are you suggesting that this if statement go?

I believe passive joints should be ignored in the [trajectory_execution_manager](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp) (which splits up trajectories and distributes them to controllers) unless they are explicitly specified to be part of a controller.

rkeatin3, 2017-12-10:
Agreed!

bmagyar, 2017-12-10:
What a happy bunch,  all agreeing! 
Where's the PR guys? :D

v4hn, 2017-12-11:
I'm waiting for Ryan's pull-request too :+1:

rkeatin3, 2017-12-12:
Alright, I'll try to find some free time to implement this in the next few days.

closed 2022-07-30 by rkeatin3