moveit/moveit #641: Test for IKFast plugin

v4hn, 2017-10-15:
Recently, we broke the ikfast plugin code in our indigo-branch and it took a while to repair. (See https://github.com/ros-planning/moveit/issues/584)

This could have been prevented by a test, but we don't have one for this yet.
Because the `ikfast_kinematics_plugin` in `moveit_kinematics` is actually a *template to generate* an ik plugin, this is not simple either.

We need a method that tests whether the current template code can be used to generate a functional ik plugin.

To tackle this, it is very much recommended that you generated an ikfast plugin before and have a general understanding of the ROS/catkin test infrastructure.

fsuarez6, 2017-10-19:
I’ll work on this

fsuarez6, 2017-10-19:
@davetcoleman I was thinking to create this test using the [PR2 robot](https://github.com/PR2/pr2_common/tree/kinetic-devel/pr2_description). What do you think?
Would [this issue](https://github.com/PR2/pr2_common/issues/265) affect the test for the IKFast plugin?

fsuarez6, 2017-10-19:
I'll try with the PR2 robot in [moveit_resources](https://github.com/ros-planning/moveit_resources)

davetcoleman, 2017-10-19:
> I'll try with the PR2 robot in moveit_resources

Perfect. There are already good examples in the moveit codebase using PR2 as a test, see https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/robot_state/test/test_kinematic_complex.cpp - it references moveit_resources

fsuarez6, 2017-10-19:
I got this working using `fanuc_description` and `fanuc_moveit_config` (available at [moveit_resources](https://github.com/ros-planning/moveit_resources)).
The test should go like this:
```bash
# Convert urdf to dae
rosrun collada_urdf urdf_to_collada `rospack find moveit_resources`/fanuc_description/urdf/fanuc.urdf fanuc.dae
# Round collada to 5 significant figures
rosrun moveit_kinematics round_collada_numbers.py fanuc.dae fanuc.rounded.dae 5
# Generate IKFast solution
python `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py --robot=fanuc.rounded.dae --iktype=transform6d --baselink=0 --eelink=8 --savefile=fanuc_manipulator_ikfast_solver.cpp
# Create dummy packages fanuc_moveit_config and  
cd ~/catkin_ws/src
catkin create pkg fanuc_moveit_config
catkin create pkg fanuc_moveit_plugins
# Copy the config folder from moveit_resources to the moveit_config pkg.
# This is moveit_config package is required by the create_ikfast_moveit_plugin.py script
cp -r moveit_resources/fanuc_moveit_config/config fanuc_moveit_config
# Build so that the two new package can be found
cd ~/catkin_ws
catkin build
# Generate the IKFast-moveit plugin
rosrun moveit_kinematics create_ikfast_moveit_plugin.py fanuc manipulator fanuc_moveit_plugins fanuc_manipulator_ikfast_solver.cpp
# Compile everything. If this succeeds, it means the IKFast plugin is OK
cd ~/catkin_ws
catkin build
```
Now the tricky part. How should we include this 'test'?
My suggestion is to include these commands directly into the travis script. What do you guys think?

Moreover, for this to work `openrave` must be installed which will take ~17 minutes. See [this travis](https://travis-ci.org/crigroup/openrave-installation) for your reference. This may be something to take into consideration given the travis [build timeouts](https://docs.travis-ci.com/user/customizing-the-build/#Build-Timeouts):

> Travis CI has specific time limits for each job, and will stop the build and add an error message to the build log in the following situations:
> - A job on travis-ci.org takes longer than 50 minutes

gavanderhoorn, 2017-10-19:
@fsuarez6: nice set of commands.

You can probably avoid building the workspace the first time by just adding `~/catkin_ws/src` to the `ROS_PACKAGE_PATH` manually. So something like:

```bash
export ROS_PACKAGE_PATH=$HOME/catkin_ws/src:$ROS_PACKAGE_PATH
```

`create_ikfast_moveit_plugin.py` should then be able to find the package (as `rospkg` should be able to find it).

> Moreover, for this to work openrave must be installed which will take ~17 minutes

Personally I never install OpenRAVE any more. I use [personalrobotics/ros-openrave](https://hub.docker.com/r/personalrobotics/ros-openrave). Last few builds haven't worked, but that may not matter for this test.

v4hn, 2017-10-19:
I believe we should definitely put these commands into a script to store with ikfast or with moveit_resources.
If it is too hard/impossible to add this to rostest, it is an option to add it to the travis script dir instead.

Thank you so much for your support @fsuarez6 !

v4hn, 2017-12-18:
@davetcoleman I consider this to be your turf unless you disagree.
We really need this test online... :)

rhaschke, 2019-06-05:
This feature was finally realized as part of #1320.

closed 2019-06-05 by rhaschke