moveit/moveit #642: Trajectory validation methods

v4hn, 2017-10-16:
Currently, we have methods in MoveIt to check [a single robot state for collision](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h#L375) and to check whether [a step between consecutive robot states is reasonable](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L1496), but we do not yet have a C++ interface to check *a full trajectory* right away.
Currently, we have methods in MoveIt to check a single robot state for collision and to check whether a step between consecutive robot states is reasonable, but we do not yet have a C++ interface to check a full trajectory passed in as an argument.

Such a method could live in https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/trajectory_processing/include/moveit/trajectory_processing/trajectory_tools.h unless you find a more fitting location for it.

Bonus points for

- adding an optional interface to check for additional constraints, see [here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h#L805)
- Wrapping the new function into a `MoveGroupCapability` to check the trajectory w.r.t. the current planning scene

jrgnicho, 2017-10-16:
@v4hn there's an ```isPathValid``` method in the [planning scene](http://docs.ros.org/jade/api/moveit_core/html/classplanning__scene_1_1PlanningScene.html#a74c475ddc55b6ba77ab9af652c81a505) that does just that.  However, it only checks for validity at each discrete joint trajectory point so any collisions between two consecutive points will not be detected,  [see here](https://github.com/ros-planning/moveit/issues/29) for a related discussion.

v4hn, 2017-10-16:
Right you are, @jrgnicho . I used these methods myself before, but apparently forgot about them.
This was on my list of outstanding issues that came up in my lab over the last few months...

I will keep this issue open because we are still short of a service that wraps it.
http://moveit.ros.org/documentation/concepts/ lists a `GetPlanValidity` service that does not exist. It can be implemented through a `MoveGroupCapability` on the basis of the functions you refer to and makes a nice exercise for someone to get involved with the `move_group` node.

BrettHemes, 2018-10-15:
This is interesting to my efforts.  I am generating custom joint trajectories and would like to do a sanity check on them before executing.  The `isPathValid` seems to be a solution.  I can add a planning scene object into my planning class but I believe a service would be a better fit possibly?  One concern I have is diverging planning scenes between my local scene and the "global" move group scene.  Any comments?

BrettHemes, 2018-10-15:
Also, it seems validity is a combination of collision and "feasibility" checking.  What exactly is feasibility?

BrettHemes, 2018-10-16:
Does feasibility capture velocity bounds by chance?

v4hn, 2018-11-26:
No, the concept of "feasibility" is not defined within moveit.
It just [verifies against some predicate the user provided](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h#L805).