moveit/moveit #643: tests for `PlanningScene`

v4hn, 2017-10-16:
MoveIt's [`PlanningScene`](https://github.com/ros-planning/moveit/tree/kinetic-devel/moveit_core/planning_scene) is at the core of many of the operations the framework performs.

Even so, the [package's tests](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/test/test_planning_scene.cpp) contains only **three** tests.

This needs to be extended to account for the many interfaces this class supports!

You don't have to add tests for *everything*.
But if you could contribute reasonable tests for *some* things, this would help a lot.

AndriyPt, 2017-10-18:
The London team (Shadow + UCL) will be working on this :)

v4hn, 2017-10-18:
Cool! Looking forward to meaningful tests :-)

felixvd, 2018-09-06:
Is this done?

AndriyPt, 2018-09-06:
@felixvd I have managed to add small amount of tests. Maybe it will make sense to keep it open till next MoveIt day :)