moveit/moveit #644: PlanningScene getMsg interfaces

v4hn, 2017-10-16:
The `PlanningScene` class has [a number of `process*` methods](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h#L707) to pass in ROS messages.

However, the only way to *extract* ROS messages from the scene is via [`getPlanningSceneMsg`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h#L686).

We would like to be able to extract individual (attached?) collision objects, the collision world, and the octomap too, via public interfaces and without the use of a `PlanningSceneMsg`.

See also [these private helper functions](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h#L940) that might be removed/replaced by this.

velveteenrobot, 2017-10-18:
Working on this.

v4hn, 2017-11-24:
fixed.

closed 2017-11-24 by v4hn