moveit/moveit #645: Validate existing moveit tutorials

giusebar, 2017-10-17:
It would be nice to check whether the existing getting started or advanced tutorials for moveit are working properly.
The tutorials could be followed and corrected if necessary.

http://docs.ros.org/kinetic/api/moveit_tutorials/html/index.html

v4hn, 2017-10-17:
You beat me to it, I just wanted to create the very same issue.

This can be chosen by multiple users (especially MoveIt!-newcomers).
If you find *anything* you don't understand without other input, please raise an issue (and if possible file a pull-request).

schntgai, 2017-10-19:
May I be assigned this issue please?

a-price, 2017-11-15:
The documentation could also use a higher-level overview describing how the tutorials fit together. For instance, the diagrams found over at http://moveit.ros.org/documentation/concepts/ would be very helpful if included in http://docs.ros.org/kinetic/api/moveit_tutorials/html/index.html. Otherwise it's not quite clear what the differences are between the planning_scene interface, move_group interface, C++ API, ROS API, etc.

davetcoleman, 2017-11-15:
+1! hopefully we will be addressing this very soon, there are plans in the works

wpmccormick, 2018-10-25:
Here at World MoveIt day in San Antonio from SwRI. I've new to MoveIt (and ROS in general) so this seems like a good one for me.

davetcoleman, 2018-10-25:
Great! Note that @mlautman reviewed the tutorials this year but the new melodic specially needs testing

wpmccormick, 2018-10-26:
Running Kinetic, I found 1 major issue I could not resolve, and 1 minor issue that slowed me down, and 1 trivial issue (question I could't find an answer for) ... so in reverse order:

1. Trivial - Could not find what EEF is [here](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#introspecting-trajectory-waypoints).

2. Minor - It wasn't clear that _"Panels"_ referred to the main menu, also [here](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#introspecting-trajectory-waypoints). Spend about 10 minutes looking for the Panels main menu option. Simply including the words _main menu_ might have done it for.

3. Major - Node failed to start [here](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html#getting-started) for the launch file in the 2nd shell `roslaunch moveit_tutorials move_group_interface_tutorial.launch`. 

I have a little more detail that I found trying to debug, but this may not be the right place to provide it. In short it appears to some issue with Rviz. I updated Rviz, and the problem got worse: in the 1st shell, `roslaunch panda_moveit_config demo.launch` failed to launch Rviz, as it did before the update.

davetcoleman, 2018-10-26:
> Minor - It wasn't clear that "Panels" referred to the main menu, also here. Spend about 10 minutes looking for the Panels main menu option. Simply including the words main menu might have done it for.

@peterso had the same issue but i don't think he ever made the pull request to clarify. please feel free to modify yourself as @dbc did here: https://github.com/ros-planning/moveit_tutorials/pull/242/files

davetcoleman, 2019-11-20:
@wpmccormick did you ever get your findings merged back into the tutorials?

Overall, this issue is over 2 years old and @mlautman did a ton of work on the tutorials since then. We now have a [new issue targeting current issues with the tutorials for ROS Melodic](https://github.com/ros-planning/moveit/issues/1731), so I'm closing this one.

closed 2019-11-20 by davetcoleman