moveit/moveit #646: [Meta-Ticket] We need reasonable test coverage!

v4hn, 2017-10-17:
MoveIt has a huge code base, but way to few tests to cover it.
This includes unit, as well as integration tests.
In order to maintain stable behavior for all use cases, such tests would very much help maintainers to verify new pull-requests and work on the project.

At the time of writing, `find moveit/ -name test | wc -l` lists 16 test folders, whereas the project counts 21 (non-meta) ROS packages (many of them with more than one submodule). So some packages don't contain *any* tests at all. But even those that do should be extendend to account for more of the API.

Please announce which package/subpackage you will work on if you want to help out.
If you don't know the package yet, this is a great opportunity to get to know it :-)

Related issues for specific tasks:
- #643 
- #641

bmagyar, 2017-10-18:
Great! How about also setting up a coverage tool?

davetcoleman, 2017-10-18:
The coverage tool is discussed [in this issue](https://github.com/ros-planning/moveit/issues/1)

fsuarez6, 2017-10-19:
From #23 I understand that the idea is to copy the cmake files in [coveralls-cmake](https://github.com/JoakimSoderberg/coveralls-cmake) into a `.coveralls` folder (I guess `moveit/.coveralls`). Is that right?
What about having a new ROS package, say `coveralls_cmake`, so that other ROS packages can use it?