moveit/moveit #648: Mongodb fails to connect the first time it is launched for a specific robot

beatrizleon, 2017-10-17:
### Description

When launching a robot for the first time, I get an error connecting with the mongo warehouse. 
This is the error:
```[ERROR] [1508254019.880601974, 1.002000000]: Unable to connect to the database at 'localhost:33829'. If you just created the database, it could take a while for initial setup.
[ERROR] [1508254019.880637834, 1.002000000]: Failed to connect to DB on localhost:33829
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
[moveit_warehouse_services-14] process has died [pid 11362, exit code -6, cmd /opt/ros/kinetic/lib/moveit_ros_warehouse/moveit_warehouse_services __name:=moveit_warehouse_services __log:=/home/user/.ros/log/690be884-b34f-11e7-99e2-0242ac110002/moveit_warehouse_services-14.log].
log file: /home/user/.ros/log/690be884-b34f-11e7-99e2-0242ac110002/moveit_warehouse_services-14*.log
```
I think there is something that we can do when the database is created to avoid this issue.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build? both
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Delete the `default_warehouse_mongo_db` inside your moveit config and try to roslaunch it again.

### Expected behaviour
Mongodb should start properly with no errors the first time:
```[ INFO] [1508254553.964394796]: Connecting to warehouse on localhost:33829
```

beatrizleon, 2017-10-17:
The London team (Shadow + UCL) will be working on this, if possible, tomorrow.

v4hn, 2017-10-17:
Thanks @beatrizleon !
I never encountered this myself, but I look forward to a pull-request that addresses this :-)

beatrizleon, 2018-10-24:
This was solved in this pr: https://github.com/ros-planning/moveit/pull/1008 so closing this issue now.

closed 2018-10-24 by beatrizleon