moveit/moveit #650: Best practices to expose more information of planning results?

bmagyar, 2017-10-18:
AFAIK there is no common policy on returning more information of planning but planning_time in the [MoveGroupAction](http://docs.ros.org/api/moveit_msgs/html/action/MoveGroup.html) . Recently this affected me both on CHOMP and STOMP. 

While of course there's no point in adding more explicit fields in the action result, perhaps a flexible, string-based aggregator could do the trick. Style would be similar to diagnostic_msgs. 

OR

Should this be solved for each planner individually with their own custom messages?

davetcoleman, 2017-10-18:
I've been frustrated with the lack of feedback from that as well. So you are advocating a ``KeyValue[] values`` field from [diagnostic_msgs](https://github.com/ros/common_msgs/blob/jade-devel/diagnostic_msgs/msg/DiagnosticStatus.msg) ?

bmagyar, 2017-10-20:
I'd be happy to see _some_ way of reporting back extra information. If it is preferred to not touch the existing messages, we could create something separate and agree on a policy (topic name, common things to report, etc)

v4hn, 2017-11-24:
If you are fine with a separate ROS topic for additional information,
I propose you could add a prototype implementation to the ompl planner that publishes a [diagnostic_msgs/DiagnosticArray](https://github.com/ros/common_msgs/blob/jade-devel/diagnostic_msgs/msg/DiagnosticArray.msg) with the information you are interested in.