moveit/moveit #654: Warn when /joint_state timestamps are expired

davetcoleman, 2017-10-18:
Currently when using ``move_group->getCurrentState()`` it calls the [current_state_monitor](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L265) ``waitForCurrentState()`` that will fail silently if the ``/joint_state`` topic it is listening to is expired. This happens, for example, when subscribing to a second computer (the robot) that has its internal clock different from the developer computer by more than 1 second. I saw this stump a new MoveIt! user recently. I believe the necessary example logic can be found [here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L220)

v4hn, 2017-10-18:
Uff.

highlighting @rhaschke because this is, as far as I know, his code.

davetcoleman, 2017-10-19:
We have a fix for this in the works

v4hn, 2019-09-06:
@davetcoleman What became of your fix?