moveit/moveit #659: Found empty JointState message error per every execution of plans

130s, 2017-10-18:
### Description

Ref. #283: Similar error msg reported

When running [chomp tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/chomp_interface_tutorial.html), the robot on RViz moves but the following error printed. `sensor_msgs/JointState` seems to be returned. I'm just wondering.

```
[ERROR] [1508358890.347763034]: Found empty JointState message
```

```
$ rostopic info /joint_states 
Type: sensor_msgs/JointState

Publishers: 
 * /joint_state_publisher (http://130s-p50:35333/)

Subscribers: 
 * /robot_state_publisher (http://130s-p50:43231/)
 * /move_group (http://130s-p50:35237/)

$ rostopic hz /joint_states 
subscribed to [/joint_states]
average rate: 10.000
        min: 0.100s max: 0.100s std dev: 0.00007s window: 9
average rate: 10.000
        min: 0.100s max: 0.100s std dev: 0.00009s window: 19
average rate: 9.999
        min: 0.100s max: 0.100s std dev: 0.00011s window: 29
average rate: 10.000
        min: 0.100s max: 0.100s std dev: 0.00011s window: 39
average rate: 10.000
        min: 0.100s max: 0.100s std dev: 0.00011s window: 49
average rate: 10.000
        min: 0.100s max: 0.100s std dev: 0.00012s window: 59
```

Overview of your issue here.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
```
cd catkin_ws/src
git clone git@github.com:ros-planning/moveit.git
git clone git@github.com:ros-planning/moveit_resources.git
cd catkin_ws
rosdep install --from-paths src --ignore-src -r -y
catkin b
source devel/setup.bash
roslaunch moveit_resources demo_chomp.launch
```

### Expected behaviour
No error msg above

### Backtrace or Console output

```
$ roslaunch moveit_resources demo_chomp.launch
... logging to /home/n130s/.ros/log/bbfbc5f6-b443-11e7-8cb7-c85b766a6139/roslaunch-130s-p50-25425.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

[... 188 lines cut ...]
[ INFO] [1508358890.129889968]: The following collision detectors are active in the planning scene.
[ INFO] [1508358890.129963355]: HYBRID
[ INFO] [1508358890.130020257]: Active collision detector is: HYBRID
[ INFO] [1508358890.231678802]: First coll check took 0.101475540
[ INFO] [1508358890.298328300]: Optimization took 0.168991 sec to create
[ INFO] [1508358890.337712695]: Forward kinematics took 0.039229650
[ INFO] [1508358890.343935018]: iteration: 0
[ INFO] [1508358890.346548452]: Chomp Got mesh to mesh safety at iter 0. Breaking out early.
[ INFO] [1508358890.346588615]: Chomp path is collision free
[ INFO] [1508358890.346662876]: Terminated after 1 iterations, using path from iteration -1
[ INFO] [1508358890.346689818]: Optimization core finished in 0.048309 sec
[ INFO] [1508358890.346713997]: Time per iteration 0.0483338
[ INFO] [1508358890.346738134]: Optimization actually took 0.217419 sec to run
[ INFO] [1508358890.346757006]: Output trajectory has 6 joints
[ INFO] [1508358890.347085987]: Joint 0 -3.13228
[ INFO] [1508358890.347110110]: Joint 1 0.187126
[ INFO] [1508358890.347129734]: Joint 2 -0.776032
[ INFO] [1508358890.347149942]: Joint 3 1.39333
[ INFO] [1508358890.347171088]: Joint 4 1.63805
[ INFO] [1508358890.347191507]: Joint 5 -5.34723
[ INFO] [1508358890.347211748]: Bottom took 0.000454 sec to create
[ INFO] [1508358890.347234365]: Serviced planning request in 0.219917 wall-seconds, trajectory duration is 0.000000
[ERROR] [1508358890.347763034]: Found empty JointState message
[ INFO] [1508358890.423946147]: Fake execution of trajectory
```

v4hn, 2017-11-24:
This is related to #579 and #589 .
It's likely a bug in the CHOMP-specific code.

bmagyar, 2017-11-24:
If it's not super urgent, assign it to me and I'll take a look. I'm
elbow-deep into chomp anyways :)

saurabhbansal90, 2018-01-04:
Hello v4hn,
I am not running CHOMP code, I am running a code to set a target pose.
But it always gives me error as "Found empty JointState message". 

Can you please suggest a solution for this. Please find my code as below:

pet1330, 2018-08-31:
@bmagyar, @v4hn @davetcoleman 
I was just wondering if anyone had found a solution to this issue yet? I found the same issue on ros-planning/moveit_tutorials#164

v4hn, 2018-09-01:
These error messages are related to a change in semantics we did a while ago.
The code is still fully functional.

Previously a lot of MoveIt code assumed it can take a current RobotState
and apply any `sensor_msgs::JointState` or `moveit_msgs::RobotState` as a diff to it
using the conversion functions in `moveit/robot_state/conversions.h`.

However, since the conversion function expects only valid (non-empty) messages now,
this has to be checked in every place the function is called.

If you still see the error, please provide a patch to fix the problematic call
to `jointStateToRobotState` or `robotStateMsgToRobotState`.