moveit/moveit #672: Extract point of end effector

ThadeuVBrito, 2017-10-24:
### Description

I am trying to get point of the arm tool  using MoveIt!. With this points I will create a code in C++ for motion planning the arm.
Now I am looking for ways to do this.

### Your environment
* ROS Distro: [Indigo]
* OS Version: Ubuntu 14.04

### Steps to reproduce
```
To make this environment I follow the steps:
cd ~/catkin_ws/src
git clone https://github.com/Kinovarobotics/kinova-ros.git kinova-ros
cd ~/catkin_ws
catkin_make

roslaunch j2n6s300_moveit_config demo.launch 
```

### Expected behaviour
Get a space coordinate of the manipulator.

### Actual behaviour
Trying use all features in Rviz display. But that does not tell the location of the arm tool.

### Backtrace or Console output
The number 2 is the point of interesting.

[image]

ThadeuVBrito, 2017-10-31:
I found a way, I do not know if is the best way. But works!
If it is help any people in future, just use the basic command:

`rostopic echo <name_of_topic_out_tool_pose>`

In my case, this command informs the position in XYZ and Orientation of the tool.

bmagyar, 2017-10-31:
Yep but keep in mind that this assumes that you have a topic publishing this pose. You may encounter cases when there isn't any. You can also use `tf` to get some information if it's regarding the current state.

For similar questions in the future I'd direct you to the [moveit-users mailing list](https://groups.google.com/forum/#!forum/moveit-users). We treat incoming _issues_ on github seriously and to keep it that way it's best to put questions where they belong ;)

closed 2017-10-31 by bmagyar

gavanderhoorn, 2017-10-31:
@bmagyar: I would actually suggest that this sort of questions should be posted on ROS Answers. That is the appropriate support channel.

bmagyar, 2017-10-31:
Good point!  :+1:

ThadeuVBrito, 2017-10-31:
I agree too, but another day I posted a question in ROS Answers about configuration in MoveIt! and a guy said to post here. Sorry for the inconvenience.

gavanderhoorn, 2017-10-31:
@ThadeuVBrito: can you post a link to the ROS Answer question?

gavanderhoorn, 2017-10-31:
I can imagine that *build errors* and bug reports would be redirected here. But general questions about usage and configuration should probably go to ROS Answers.

ThadeuVBrito, 2017-10-31:
Yes, I can.