moveit/moveit #673: MotionPlanResponse is always error_code.val = 0 when planning is failed

knorth55, 2017-10-25:
### Your environment
* ROS Distro: [Indigo]
* OS Version: 14.04
* Source
* 0.7.12

### Description

When `context_->planning_pipeline_->generatePlan(ps, req.motion_plan_request, mp_res);` returns `false`, `MotionPlanResponse` returns initial value such as `error_code = 0`.
That is because when service call callback function returns `false`, service returns response with initial value.
Therefore, when `solved` is `false`, `MotionPlanResponse.error_code.val` is `0` and it is confusing.
IMO, `service call` function should always return `true`, and we can get valid `error_code`.

silkjc, 2017-11-02:
+1..Nobody checking return codes before this ?

davetcoleman, 2017-11-02:
Is this an issue with how ROS implements service calls that are aborted, not a MoveIt! issue.

pHaeusler, 2017-11-02:
The service callback should return true. The response object has the appropriate error code to indicate the pipeline failed.

From [roscpp services wiki](http://wiki.ros.org/roscpp/Overview/Services)

> A return value of false means the call has failed and the response object will not be sent to the caller.

knorth55, 2017-11-03:
> Is this an issue with how ROS implements service calls that are aborted, not a MoveIt! issue.

What do you mean?
Is this a issue for `ros_comm`?

It seems MoveIt! ROS implementation issue.
Service-call callback must return true when we need to get valid response.
What I want to know is the situation when MoveIt! want to return valid response.
If Moveit! want to return valid response in any cases, service-call callback should always return `true`.

knorth55, 2017-11-09:
@davetcoleman 
Can you give me an advice about this issue?
I do not know whole structure of `moveit`, so I have no idea about side effect.
If there is no side effect, I think `computePlanService` should always return `true` for valid error code.

davetcoleman, 2017-11-09:
A pull request with your suggestion would be easier to provide feedback on, thanks!

knorth55, 2017-11-09:
I already make a PR #674

knorth55, 2017-11-27:
Solved!

closed 2017-11-27 by knorth55