moveit/moveit #676: Moveit throws error for gripper_action_controller

rfn123, 2017-10-26:
I'm not quite sure if it belongs here or to ros_control,
but I recently wanted to implement a gripper_action_controller within Moveit, and after some trying I found out that it always throws the error: 
`Name and joints must be specifed for each controller`

Looking into the code [here](https://github.com/ros-planning/moveit/blob/5c5fcd7804daf349310db0d28539010e23c51dc0/moveit_plugins/moveit_simple_controller_manager/src/moveit_simple_controller_manager.cpp#L72) tells me that this error is thrown when either '`joints`' (plural) or '`name`' is not found in the yaml file for the specific controller.

However, as stated in https://github.com/ros-controls/ros_controllers/issues/155, gripper_action_controller is only working with a single-dof gripper, therefore the controller can only be started if the .yaml param is specified as '`joint`' in the singular form.

An example of a correct yaml file for gripper_action_controller but throwing the above mentioned error would be:
```
controller_list:
 - name: rovi_uarm/gripper_controller
   action_ns: gripper_action
   type: GripperCommand
   default: true
   joint: gripper_joint
```

It's not a severe problem though as the gripper controller seems to start anyways despite the error, but it would be nice if this could get fixed somehow :)

bmagyar, 2017-10-26:
> However, as stated in ros-controls/ros_controllers#155

Hiya,

Nice catch!
Just to reiterate,  moveit is looking for the "joints"  field on the param
server as common with most ros_control controllers while the actual
controller in this case uses a single joint,  hence the param field name
'joint'?

It's a tricky question but I think it'd make sense to standardize this over
at ros_control.

rfn123, 2017-10-27:
@bmagyar Thanks for your response! Yes, at least the code from gripper [https://github.com/ros-controls/ros_controllers/blob/9f5eaef834cc61bac4992decd8baa93e5fb6372a/gripper_action_controller/include/gripper_action_controller/gripper_action_controller_impl.h#L152](url) tells me that the gripper_action_controller is looking for the param 'joint' without 's'. I think I'll open another issue in ros_control then.

ipa-mdl, 2017-10-27:
IMHO you are mxing up the configurations here.
The configuration for linking MoveIt to the actual gripper interface takes `joints`.
The configuration of the `ros_control`  controller takes `joint`
Apart from that they differ in the *content* for `type` as well, so you cannot use one for the other and vice versa.

rfn123, 2017-10-28:
@ipa-mdl You actually made a good point to distinguish the joint/joints params in the different config files. After some more testing, I realized that my problem was somewhere else though. 

for the Moveit config, the action_ns actually has to be **`gripper_cmd`** and not `gripper_action`:
```
controller_list:
 - name: gripper_controller
   action_ns: gripper_cmd
   type: GripperCommand
   default: true
   joints: 
      - gripper_joint
```

closed 2017-10-28 by rfn123