moveit/moveit #678: create move group show URDF parse error

jacknlliu, 2017-10-27:
### Description

I use the following code 
```
moveit::planning_interface::MoveGroupInterface arm_commander_("manipulator");
```

But I got the following errors about URDF file parse,
```
[ERROR] [1509083111.796501843]: Unknown geometry type ''
[ERROR] [1509083111.796576545]: Could not parse visual element for Link [base_link]
[ERROR] [1509083111.796651044]: Unknown geometry type ''
[ERROR] [1509083111.796674800]: Could not parse visual element for Link [shoulder_link]
...
Segmentation fault (core dumped)
```

but I check my `urdf` file with `check_urdf`, no errors.  And then I use rviz  moveit planning plugin also works.

So why I got the above errors?
 
Any help will be appreciated!

###  environment
* ROS Distro: kinetic
* OS Version: Ubuntu 16.04

bmagyar, 2017-10-27:
Could you also share your urdf please?

jacknlliu, 2017-10-27:
Here is my URDF file
```xml
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from /data/assembler/assembler-dynamic/src/assembler_description/urdf/assembler_robot_gazebo_world.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="assembler_robot" xmlns:xacro="http://ros.org/wiki/xacro">
  <!-- we still not wrap this as a robot name, because we should add hand -->
  <gazebo>
    <plugin filename="libgazebo_ros_control.so" name="ros_control">
      <!--robotNamespace>/</robotNamespace-->
      <!--robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType-->
    </plugin>
    <!--
    <plugin name="gazebo_ros_power_monitor_controller" filename="libgazebo_ros_power_monitor.so">
      <alwaysOn>true</alwaysOn>
      <updateRate>1.0</updateRate>
      <timeout>5</timeout>
      <powerStateTopic>power_state</powerStateTopic>
      <powerStateRate>10.0</powerStateRate>
      <fullChargeCapacity>87.78</fullChargeCapacity>     
      <dischargeRate>-474</dischargeRate>
      <chargeRate>525</chargeRate>
      <dischargeVoltage>15.52</dischargeVoltage>
      <chargeVoltage>16.41</chargeVoltage>
    </plugin>
-->
  </gazebo>
  <!-- measured from model -->
  <!--property name="shoulder_height" value="0.089159" /-->
  <!--property name="shoulder_offset" value="0.13585" /-->
  <!-- shoulder_offset - elbow_offset + wrist_1_length = 0.10915 -->
  <!--property name="upper_arm_length" value="0.42500" /-->
  <!--property name="elbow_offset" value="0.1197" /-->
  <!-- CAD measured -->
  <!--property name="forearm_length" value="0.39225" /-->
  <!--property name="wrist_1_length" value="0.093" /-->
  <!-- CAD measured -->
  <!--property name="wrist_2_length" value="0.09465" /-->
  <!-- In CAD this distance is 0.930, but in the spec it is 0.09465 -->
  <!--property name="wrist_3_length" value="0.0823" /-->
  <!-- manually measured -->
  <link name="base_link">
    <visual>
      <geometry>
        <mesh filename="package://ur_description/meshes/ur5/visual/base.dae"/>
      </geometry>
      <material name="LightGrey">
        <color rgba="0.7 0.7 0.7 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://ur_description/meshes/ur5/collision/base.stl"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="4.0"/>
      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
      <inertia ixx="0.00443333156" ixy="0.0" ixz="0.0" iyy="0.00443333156" iyz="0.0" izz="0.0072"/>
    </inertial>
  </link>
  <joint name="shoulder_pan_joint" type="revolute">
    <parent link="base_link"/>
    <child link="shoulder_link"/>
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.089159"/>
    <axis xyz="0 0 1"/>
    <limit effort="150.0" lower="-3.14159265359" upper="3.14159265359" velocity="3.15"/>
    <dynamics damping="0.0" friction="0.0"/>
[... 796 lines cut ...]
          <far>3.0</far>
        </clip>
      </camera>
      <plugin filename="libgazebo_ros_openni_kinect.so" name="kinect_camera_controller">
        <cameraName>3dcamera</cameraName>
        <alwaysOn>true</alwaysOn>
        <updateRate>20</updateRate>
        <imageTopicName>rgb/image_raw</imageTopicName>
        <depthImageTopicName>depth/image_raw</depthImageTopicName>
        <pointCloudTopicName>depth/points</pointCloudTopicName>
        <cameraInfoTopicName>rgb/camera_info</cameraInfoTopicName>
        <depthImageCameraInfoTopicName>depth/camera_info</depthImageCameraInfoTopicName>
        <frameName>camera_depth_optical_frame</frameName>
        <baseline>0.1</baseline>
        <distortion_k1>0.0</distortion_k1>
        <distortion_k2>0.0</distortion_k2>
        <distortion_k3>0.0</distortion_k3>
        <distortion_t1>0.0</distortion_t1>
        <distortion_t2>0.0</distortion_t2>
        <pointCloudCutoff>0.4</pointCloudCutoff>
      </plugin>
    </sensor>
  </gazebo>
</robot>
```

jacknlliu, 2017-10-27:
More details, I can use `gazebo` load my scene correctly, and use rviz `moveit planning plugin` to make my robot move as expected.

Only with the move group code, I got above errors.

bmagyar, 2017-10-27:
I see. Interesting...On first read I couldn't spot any obvious problems. Try to get deeper and run `move_group` in debug mode to get to the exact location of the problem.

jacknlliu, 2017-10-27:
@bmagyar  I found that this related a function we need statically linked to the final executable application, but we never use it in our main function which only includes one line substance code with moveit API. 

I make an [unit test ros package](https://github.com/jacknlliu/moveit_unit_test) which can directly build and test. In this package, I just add a class which we haven't used it. I confirm if we remove the [linked function](https://github.com/jacknlliu/moveit_unit_test/blob/master/src/HapticDevice.cpp#L10) which I actually haven't used will not make trouble for moveit api. 

What makes moveit trouble?

bmagyar, 2017-10-28:
The symptoms are way too random to be a problem with MoveIt. 

Your trouble is probably with those statically linked libraries. I'd advise analysing the binary dependencies of both move group and your libraries with `nm` or you preferred symbol reader and see if there's a coincidence. They might be trying to access the same library but different versions. This happens often with boost for instance.

jacknlliu, 2017-10-28:
@bmagyar  Thanks for your suggestion, I will try it. But we should know that the linked function not used in our main function, why moveit API can't work. Actually they don't have same dependencies, and the linked function should not affect moveit.

 I have test them in different ros package and they can work very well, just can't make them together. Do you have some suggestions about statically link an object using [CMakelists.txt](https://github.com/jacknlliu/moveit_unit_test/blob/master/CMakeLists.txt#L61)? I don't know if there is a problem.

jacknlliu, 2017-10-29:
It's too much work to do if I parse so many and deep dependencies manually. If we don't have more suggestions,  I will close this issue...

closed 2017-10-31 by jacknlliu