moveit/moveit #680: moveit not work with mimic joints

jacknlliu, 2017-11-06:
We use moveit to plan a group state of [hand](https://github.com/RobotnikAutomation/barrett_hand_common/blob/indigo-devel/barrett_hand_description/urdf/bh282.urdf.xacro#L158) which including mimic joints, the plan failed to execute.  

```
[ WARN] [1509951782.181869557, 1706.266000000]: Fail: ABORTED: No motion plan found. No execution attempted.
[ INFO] [1509951782.182363669, 1706.266000000]: ABORTED: Solution found but controller failed during execution
```

```
More than 80% of the sampled goal states fail to satisfy the constraints imposed on the goal sampler. Is the constrained sampler working correctly?

[ INFO] [1509951782.171828838, 1706.256000000]: Constraint satisfied:: Joint name: 'bh_j11_joint', actual value: 0.000027, desired value: 0.000000, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1509951782.171852837, 1706.256000000]: Constraint satisfied:: Joint name: 'bh_j12_joint', actual value: 1.712097, desired value: 1.712000, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1509951782.171877275, 1706.256000000]: Constraint violated:: Joint name: 'bh_j13_joint', actual value: 0.589410, desired value: 0.046400, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1509951782.171893954, 1706.256000000]: Constraint satisfied:: Joint name: 'bh_j21_joint', actual value: 0.000027, desired value: 0.000000, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1509951782.171909929, 1706.256000000]: Constraint satisfied:: Joint name: 'bh_j22_joint', actual value: 1.590696, desired value: 1.590700, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1509951782.171930821, 1706.256000000]: Constraint violated:: Joint name: 'bh_j23_joint', actual value: 0.547617, desired value: 0.454800, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1509951782.171959565, 1706.256000000]: Constraint satisfied:: Joint name: 'bh_j32_joint', actual value: 1.442321, desired value: 1.442400, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1509951782.171978114, 1706.256000000]: Constraint violated:: Joint name: 'bh_j33_joint', actual value: 0.496537, desired value: 0.227400, tolerance_above: 0.000100, tolerance_below: 0.000100
[ERROR] [1509951782.181404258, 1706.265000000]: RRTConnect: Unable to sample any valid states for goal tree
```

Ridhwanluthra, 2018-06-09:
@jacknlliu Were you able to find a solution?

jacknlliu, 2018-06-09:
I think this is still an issue.

Ridhwanluthra, 2018-06-09:
are you able to plan using the rviz interface?

jacknlliu, 2018-06-10:
I think you can't use the mimic joints currently. And then you can plan with these joints, but not rviz, just moveit codes what you posted there.

v4hn, 2018-06-11:
Planning to reach "open" / "closed" joint states works for the PR2 gripper, which includes mimic joints `l_gripper_l_finger_joint` / `l_gripper_r_finger_joint`.

So whatever problem you hit @jacknlliu , it's not simply a matter of "mimic joints don't work".

jacknlliu, 2018-06-11:
@v4hn  Thanks for your attention, could you give the link address of PR2 gripper using moveit? I will have a try for this. Actually I can't use moveit to plan the mimic joints.

v4hn, 2018-06-11:
Turns out I referred to our (currently non-public) configuration of our PR2.

The same should work with the upstream configuration [here](https://github.com/ros-planning/moveit_pr2/). The description should be released in kinetic as `ros-kinetic-pr2-description` or can be found [here](https://github.com/PR2/pr2_common).

You have to add the named states by hand in the srdf though:

```
    <group_state name="open" group="left_gripper">
        <joint name="l_gripper_joint" value="0.088" />
        <joint name="l_gripper_motor_screw_joint" value="1.0" />
        <joint name="l_gripper_l_finger_joint" value="0.477" />
        <joint name="l_gripper_r_finger_joint" value="0.477" />
        <joint name="l_gripper_r_finger_tip_joint" value="0.477" />
        <joint name="l_gripper_l_finger_tip_joint" value="0.477" />
    </group_state>
    <group_state name="closed" group="left_gripper">
        <joint name="l_gripper_joint" value="0.0" />
        <joint name="l_gripper_motor_screw_joint" value="0.0" />
        <joint name="l_gripper_l_finger_joint" value="0.002" />
        <joint name="l_gripper_r_finger_joint" value="0.002" />
        <joint name="l_gripper_r_finger_tip_joint" value="0.002" />
        <joint name="l_gripper_l_finger_tip_joint" value="0.002" />
    </group_state>
```

jacknlliu, 2019-01-16:
@rhaschke I find it works in our recent development.

closed 2019-01-16 by jacknlliu