moveit/moveit #681: KDL Plugin not found after installing Trac_IK

vighv, 2017-11-06:
I installed the Trac_IK package to use with moveit. 
But now when I try to switch back to KDL, I get the following error:

```
[ERROR] [1510011662.923254441]: The kinematics plugin (arm) failed to load. Error: According to the
 loaded plugin descriptions the class kdl_kinematics_plugin/KDLKinematicsPlugin with base class type 
kinematics::KinematicsBase does not exist. Declared types are  
baxter_left_arm_kinematics/IKFastKinematicsPlugin 
baxter_right_arm_kinematics/IKFastKinematicsPlugin pr2_arm_kinematics/PR2ArmKinematicsPlugin 
trac_ik_kinematics_plugin/TRAC_IKKinematicsPlugin
```

Environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04

gavanderhoorn, 2017-11-07:
The KDL plugin is part of MoveIt, so it should be present, provided you have the `moveit_kinematics` package installed.

How did you install MoveIt and Trac Ik? From `deb`s or a source build?

v4hn, 2017-11-24:
As @gavanderhoorn already said, you don't have `moveit_kinematics` (ros-indigo-moveit-kinematics) installed and you also miss other plugins from there.

I will close this as invalid.

closed 2017-11-24 by v4hn