moveit/moveit #683: self-collision padding not configurable

simonschmeisser, 2017-11-12:
In https://github.com/ros-planning/moveit_core/pull/168 there was a discussion about adding self-collision paddings between ie the actual robot and fixed parts of the scene. We ran into some cases where this would be really useful/necessary (UR5, yumi) due to imprecise movements. It wouldn't hurt with other robots as well as people would be more relaxed not seeing it go by very closely to ie a edge of a table. I guess this could also be achieved using oversized collision geometries but that feels a bit hackish.

See my unanswered question here 
https://answers.ros.org/question/274892/padding-between-robot-and-fixed-environment/ as well

So is there any interest in a PR for a separate padding category for self-collisions?

bmagyar, 2017-11-12:
> In ros-planning/moveit_core#168
> environment/ as well

Big thumbs up.

 I had this with humanoids a few years back. The solution was oversized
meshes which was a big hack. If I remember well arm_navigation used to
support this.

davetcoleman, 2017-11-12:
This would be a great contribution, but you need to be careful where that padding is applied throughout the code base. In some cases you do not want to use it. For example, the robot's current state might not be in collision unless you are using the padding. But if MoveIt! thinks the current state is in collision, it will not be able to plan (except it does have a "wiggle joints" algorithm that will try to find a nearby state that is not in collision).

v4hn, 2017-12-05:
> So is there any interest in a PR for a separate padding category for self-collisions?

YES!

sjhansen3, 2018-02-12:
Is there any update on this? I dont understand the following based on the several posts I have read:
 
1. Is there a way to add padding to the links of the robot?
2. Is there a way to add padding to scene objects?
3. Is there a way to add padding to anything with our without YAML files?

[(https://github.com/ros-planning/moveit_ros/issues/341)]
seems to indicate that it might be possible to add padding, but Its still unclear. I tried a few different name spaces to no avail.

Sorry for my confusion, but after reading every ROS answer on this topic I am thoughrouly confused :/

Bottom line: What features regarding padding do and do not exist currently?

Thanks!

jayalatn, 2020-09-21:
Is someone working on self collision padding atm or is it already added ?

v4hn, 2020-09-21:
> Is someone working on self collision padding atm or is it already added?

[peci1 spent a lot of time fixing issues with the underlying `geometric_shapes` package](https://github.com/ros-planning/geometric_shapes/pulls?q=is%3Apr+author%3Apeci1), but to the best of my knowledge nobody worked on better higher-level integration in MoveIt.

jayalatn, 2020-09-22:
thanks for the update.