moveit/moveit #686: New release 201711

130s, 2017-11-13:
New release is planned on the weekend of Nov 18, 19.
As usual, please send any concern, wait list.
Also, since this is going to be the 1st post-WMD release, (esp. maintainers) please try pushing the open PRs if any.

v4hn, 2017-11-14:
Uff, slave-driver 😱 

Could you please postpone the planned release by one week to give me some more time to work through the open requests?
As you noticed we still have way too many things open at the moment.

130s, 2017-11-15:
Ok, then 25-26 weekend.

I didn't mean to sound pushy...I just thought we needed a set timeline. Hope the new date works better, and of course no need for all WMD efforts to be released by this release.

v4hn, 2017-11-15:
Sorry, the previous comment was meant to sound more funny than it does...
I fully agree that we should do a release soon.
I'm just a bit behind schedule here and would like to catch up before we release.

25-26th sounds ok.

bmagyar, 2017-11-15:
Dear slave-drivers,

Can I add back CHOMP to the released-installed stack once I finish adding the tests? 
#626 is waiting only small changes to be added and a review (which could already start [ wink wink @v4hn  ] ). I can try to grind through the last bits this weekend.

Btw I think we have a friendly-enough atmosphere here to allow pushy and rude things to be said ;)

130s, 2017-11-27:
I'm trying to make a release with `bloom` then hit a problem https://github.com/ros-infrastructure/bloom/issues/448. I can technically release without it but it's way convenient and less error-prone with it. So unless there's strong demand I'd like to wait for that issue to be resolved (hopefully sometime this week).

v4hn, 2017-11-28:
Should be fine. Go ahead with the release whenever bloom is fixed (once again ;))

130s, 2017-12-10:
Kinetic 0.9.10 https://github.com/ros/rosdistro/pull/16566
Lunar 0.9.10 https://github.com/ros/rosdistro/pull/16567

For Indigo I'm making a set PRs that seem to have not been cherrypicked #715.

v4hn, 2017-12-10:
Oh, I know we were waiting for you to make the release for a few days now,
but that still came a bit as a surprise :-)
Thank you for doing it!

We realized on Friday that there were still two shortcomings in #701, that make it harder to use it with existing kinematics plugins and henhenhen filed #714 to adjust them.
I would like to have a release **including** #714 in the next publicly synced release of indigo and kinetic.
Otherwise it will take even longer to finally get rid of this bug, because we can only fix them in individual external plugins once #714 is released and synced in indigo and kinetic...
@130s Could I ask you to make a followup release once #714 got merged?
Of course you are also very much welcome to review it so that we can move on right away. :D

130s, 2017-12-18:
For [Kinetic 0.9.10](https://discourse.ros.org/t/new-packages-for-kinetic-2017-12-11/3378) is already out publicly.

For KL, I can make a quick release as requested (0.9.11 for both). For Indigo, new release hasn't been even made and I can make a release where the corresponding changes to both 0.9.10 and 0.9.11 will be included. It'll be easier if someone sends a queue on this ticket.

rhaschke, 2017-12-18:
@130s There is another regression in the latest Kinetic/Lunar release, which is fixed (but not yet merged) in #724.

130s, 2017-12-25:
Indigo https://github.com/ros/rosdistro/pull/16652
Kinetic https://github.com/ros/rosdistro/pull/16654
Lunar https://github.com/ros/rosdistro/pull/16653

closed 2017-12-25 by 130s