moveit/moveit #687: Unable to sample any valid states for goal tree with RRTConnectkConfigDefault planner

aravindsiv, 2017-11-14:
### Description
I am trying to use MoveIt! on my KUKA LBR iiwa 14 R820 with [this](https://github.com/SalvoVirga/iiwa_stack) package being used to interface with the robot controllers, publish the robot state, etc. When I try to plan for a 6D pose whose orientation is parallel to one of the robot axes (say, quaternion q = [0,1,0,0]), I get the following error with the RRTConnectkConfigDefault planner. I'm also making use of the TRAC-IK Kinematics Plugin for MoveIt!. 

A couple of notes:

* I am able to plan to practically any 3D pose in my workspace, as long as I preserve the previous configuration's orientation.
* I am able to manually move my robot to the same pose, with some margin of error (+/- 2cm, +/- 2 degrees).

### My environment
* ROS Distro: Indigo
* Ubuntu 14.04
* Binary build, version 0.7.12-0trusty-20170915-202331-0800 (?)

### Backtrace or Console output

```
[ INFO] [1510693932.560814655]: Handling joint: iiwa_joint_1
[ INFO] [1510693932.560980141]: Handling joint: iiwa_joint_2
[ INFO] [1510693932.561025295]: Handling joint: iiwa_joint_3
[ INFO] [1510693932.561056911]: Handling joint: iiwa_joint_4
[ INFO] [1510693932.561090225]: Handling joint: iiwa_joint_5
[ INFO] [1510693932.561121215]: Handling joint: iiwa_joint_6
[ INFO] [1510693932.561152539]: Handling joint: iiwa_joint_7
[ INFO] [1510693932.561189623]: Register state and effort interfaces
[ INFO] [1510693932.583748106]: Loading robot model 'iiwa14'...
[ INFO] [1510693932.583816557]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1510693932.678966042]: IK Using joint iiwa_link_1 -2.93215 2.93215
[ INFO] [1510693932.679039856]: IK Using joint iiwa_link_2 -2.05949 2.05949
[ INFO] [1510693932.679073995]: IK Using joint iiwa_link_3 -2.93215 2.93215
[ INFO] [1510693932.679103322]: IK Using joint iiwa_link_4 -2.05949 2.05949
[ INFO] [1510693932.679133296]: IK Using joint iiwa_link_5 -2.93215 2.93215
[ INFO] [1510693932.679148802]: IK Using joint iiwa_link_6 -2.05949 2.05949
[ INFO] [1510693932.679179867]: IK Using joint iiwa_link_7 -3.01942 3.01942
[ INFO] [1510693932.679210053]: Looking in private handle: /iiwa/move_group for param name: manipulator/position_only_ik
[ INFO] [1510693932.679783253]: Looking in private handle: /iiwa/move_group for param name: manipulator/solve_type
[ INFO] [1510693932.680243809]: Using solve type Speed
[INFO] [WallTime: 1510693932.864263] Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [WallTime: 1510693932.865361] Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [WallTime: 1510693932.866385] Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [WallTime: 1510693932.867428] Loading controller: joint_state_controller
[ INFO] [1510693932.890871032]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1510693932.897064834]: MoveGroup debug mode is OFF
Starting context monitors...
[ INFO] [1510693932.897102514]: Starting scene monitor
[INFO] [WallTime: 1510693932.898399] Loading controller: PositionJointInterface_trajectory_controller
[ INFO] [1510693932.899397209]: Listening to '/iiwa/planning_scene'
[ INFO] [1510693932.899417791]: Starting world geometry monitor
[ INFO] [1510693932.907685524]: Listening to '/iiwa/collision_object' using message notifier with target frame '/world '
[ INFO] [1510693932.909657679]: Listening to '/iiwa/planning_scene_world' for planning scene world geometry
[INFO] [WallTime: 1510693932.939216] Controller Spawner: Loaded controllers: joint_state_controller, PositionJointInterface_trajectory_controller
[INFO] [WallTime: 1510693932.941190] Started controllers: joint_state_controller, PositionJointInterface_trajectory_controller
[ INFO] [1510693932.985533714]: Listening to '/iiwa/attached_collision_object' for attached collision objects
Context monitors started.
[ INFO] [1510693933.023450394]: Initializing OMPL interface using ROS parameters
[ INFO] [1510693933.037785413]: Using planning interface 'OMPL'
[ INFO] [1510693933.060738197]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1510693933.061188363]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1510693933.061547281]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1510693933.061886849]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1510693933.062202846]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1510693933.062536395]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1510693933.062571653]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1510693933.062583596]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1510693933.062593297]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1510693933.062603502]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1510693933.062613588]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1510693933.086582208]: Loading robot model 'iiwa14'...
[ INFO] [1510693933.086615768]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1510693933.204702422]: IK Using joint iiwa_link_1 -2.93215 2.93215
[ INFO] [1510693933.204762471]: IK Using joint iiwa_link_2 -2.05949 2.05949
[... 50 lines cut ...]
[ INFO] [1510693940.160161651]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ERROR] [1510693940.160435658]: Found empty JointState message
[ERROR] [1510693940.160521092]: Found empty JointState message
[ERROR] [1510693940.160867626]: Found empty JointState message
[ERROR] [1510693940.161946586]: Found empty JointState message
[ INFO] [1510693940.163882295]: No optimization objective specified, defaulting to PathLengthOptimizationObjective
[ INFO] [1510693940.164021761]: Planner configuration 'manipulator[RRTConnectkConfigDefault]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1510693940.165740395]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1510693940.165801366]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1510693940.165919486]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1510693940.165986591]: manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ERROR] [1510693941.172900052]: manipulator[RRTConnectkConfigDefault]: Unable to sample any valid states for goal tree
[ INFO] [1510693941.172966059]: manipulator[RRTConnectkConfigDefault]: Created 1 states (1 start + 0 goal)
[ERROR] [1510693941.173111390]: manipulator[RRTConnectkConfigDefault]: Unable to sample any valid states for goal tree
[ERROR] [1510693941.173194763]: manipulator[RRTConnectkConfigDefault]: Unable to sample any valid states for goal tree
[ INFO] [1510693941.173277033]: manipulator[RRTConnectkConfigDefault]: Created 1 states (1 start + 0 goal)
[ERROR] [1510693941.173641889]: manipulator[RRTConnectkConfigDefault]: Unable to sample any valid states for goal tree
[ INFO] [1510693941.173697575]: manipulator[RRTConnectkConfigDefault]: Created 1 states (1 start + 0 goal)
[ INFO] [1510693941.173770942]: manipulator[RRTConnectkConfigDefault]: Created 1 states (1 start + 0 goal)
[ WARN] [1510693941.173893245]: ParallelPlan::solve(): Unable to find solution by any of the threads in 1.008947 seconds
[ INFO] [1510693942.177730249]: Unable to solve the planning problem
[ WARN] [1510693942.178190857]: Fail: ABORTED: No motion plan found. No execution attempted.
[ INFO] [1510693942.180097650]: Received new trajectory execution service request...
[ WARN] [1510693942.180181189]: The trajectory to execute is empty
```

gavanderhoorn, 2017-11-15:
> say, quaternion q = [0,1,0,0]

Just making sure: that is an actual quaternion, right? Not an RPY triplet?

Qifan94, 2017-11-16:
Hi, I have the same problem now. Could you tell me if you fix it out? 
Thanks!
Email: qwang94@163.com

aravindsiv, 2017-11-16:
Figured this out. Turns out, [0,1,0,0] is a valid quaternion, but doesn't have the required precision necessary for TRAC-IK to return a valid IK. Instead, I'd recommend using ROS's ```tf.transformations``` method to construct the quaternion for your target pose.

closed 2017-11-16 by aravindsiv

sritee, 2019-12-09:
I Tried the following things and one of them got it work 

Increased goal-setting tolerance to 0.05
PoseMaybe not reachable, perhaps x/y/z out of reach. 
The Quarternion set was maybe not feasible. Instead took a feasible quaternion by echoing the pose.

mgagnier93, 2020-05-13:
Hi all!

My motion planner works when its not around parts (very useful!) but when the robot gripper gets close to any surfaces, it isn't able to find a valid path, and I get the same issue that everyone is discussing. 

I have tried increasing the computation time to 60s without any better results. I also tried changing the IK solver to "LBKPIECE1", "SBLkConfigDefault", and "RRTConnectkConfigDefault". I also tried giving larger position and orientation tolerances ( e.g. move_group.set_goal_orientation_tolerance(0.5)) which did not help - it either still could not find a solution, or the solution was to a point that was also not close to the surface. I also tried specifying my quaternions using tf (e.g. quaternion = tf.transformations.quaternion_from_euler(pi, 0.0, 0.0)).

The weird thing is that if I open up RVIZ and manually drag the robot to a slightly lower point, the RVIZ GUI is able to find a path and execute it. 

Attached is a gazebo snapshot of the robot reaching as far down as it can before it can't find a path anymore. I have collisions displayed, and as you can see there's nothing preventing the robot from reaching farther down. 

I've attached my code to control the robot which is essentially the code from the tutorial below with the unnecessary code deleted. 

http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html#start-rviz-and-movegroup-node 

[open_box_tester.txt](https://github.com/ros-planning/moveit/files/4619230/open_box_tester.txt)

[image]