moveit/moveit #690: RViz MotionPlanner plugin: broken manipulator geometry

dbasa, 2017-11-16:
### Description

I created a MoveIt! configuration for a manipulator with 3 DOF. When I launch the demo.launch file from the newly created configuration package the geometry of my manipulator is broken. Besides the fact that the primitives are not visible (meshes are visible), in `Scene Robot` each link's and each joint's position is `[0;0,0]` and orientation seems random (but reproducible). When I add the `RobotModel` plugin the links are aligned correctly, but joints are still skewed.

When running rviz I see the following Errors/Infos:

```
[ INFO] [1510830451.608022093]: No active joints or end effectors found for group ''. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1510830451.608246183]: No active joints or end effectors found for group 'e'. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1510830451.609597982]: No active joints or end effectors found for group 'e'. Make sure you have defined an end effector in your SRDF file and that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1510830451.609934654]: Constructing new MoveGroup connection for group 'e' in namespace ''
[ INFO] [1510830452.655901987]: Ready to take commands for planning group e.
```

Here is my simplified the URDF file:

```
<?xml version="1.0" ?>
<robot name="robot">

  <link name="footprint"/>

  <joint name="p_joint" type="fixed">
    <parent link="footprint"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <child link="base"/>
  </joint>

  <link name="base">
    <visual name="visual">
      <origin rpy="0 1.57079632679  0" xyz="0.035 0 0"/>
      <geometry>
        <cylinder length="0.07" radius="0.065"/>
      </geometry>
    </visual>
    <inertial>
      <origin rpy="0 0 0" xyz="0.035 0 0"/>
      <mass value="0.5"/>
      <inertia ixx="0.000732291666667" ixy="0" ixz="0" iyy="0.000732291666667" iyz="0" izz="0.00105625"/>
    </inertial>
  </link>

  <joint name="q0" type="revolute">
    <parent link="base"/>
    <child link="rotatory"/>
    <origin rpy="1.57079632679 0 1.57079632679" xyz="0.07 0 0"/>
    <axis xyz="0 0 1"/>
    <!-- joint range >= 2*PI leads to problems at least in simulation -->
    <limit effort="30" lower="-3.11017672705" upper="3.11017672705" velocity="0.314159265359"/>
  </joint>

  <transmission name="manipulator_swivel_trans">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="q0">
      <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
    </joint>
    <actuator name="manipulator_swivel_motor">
      <mechanicalReduction>100</mechanicalReduction>
    </actuator>
  </transmission>

  <link name="rotatory">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0.05535"/>
      <geometry>
        <cylinder length="0.117" radius="0.055"/>
      </geometry>
    </visual>
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0.05"/>
      <mass value="0.5"/>
      <inertia ixx="0.00155516666667" ixy="0" ixz="0" iyy="0.00155516666667" iyz="0" izz="0.00105625"/>
    </inertial>
  </link>

  <joint name="q1" type="revolute">
    <parent link="rotatory"/>
    <child link="lever_arm"/>
    <origin rpy="1.57079632679 0 -1.57079632679" xyz="0 0 0.117"/>
    <axis xyz="0 0 1"/>
    <limit effort="30" lower="-0.5934" upper="0.8203" velocity="0.314159265359"/>
  </joint>

  <transmission name="leverarm_trans">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="q1">
      <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
    </joint>
    <actuator name="leverarm_motor">
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>

  <link name="lever_arm">
    <visual>
      <origin rpy="-1.57079632679 0 0" xyz="0 0.12625 0"/>
      <geometry>
        <cylinder length="0.2525" radius="0.045"/>
      </geometry>
    </visual>
    <inertial>
      <origin rpy="0 0 0" xyz="0 0.12625 0"/>
      <mass value="5"/>
      <inertia ixx="0.106260416667" ixy="0" ixz="0" iyy="0" iyz="0" izz="0.106260416667"/>
    </inertial>
  </link>

  <joint name="q2" type="revolute">
    <parent link="lever_arm"/>
    <child link="pointer"/>
    <origin rpy="0 0 1.57079632679" xyz="0 0.2525 0"/>
    <axis xyz="0 0 1"/>
    <limit effort="30" lower="-1.57079632679" upper="1.57079632679" velocity="0.314159265359"/>
  </joint>

  <transmission name="cutter_swivel_trans">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="q2">
      <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
    </joint>
    <actuator name="cutter_swivel_motor">
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>

  <link name="pointer">
    <visual>
      <origin rpy="0 1.57079632679 0" xyz="0.04 0 0"/>
      <geometry>
        <cylinder length="0.08" radius="0.035"/>
      </geometry>
    </visual>
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <mass value="3.3"/>
      <inertia ixx="0.00428615" ixy="0" ixz="0" iyy="0.005253875" iyz="0" izz="0.005931475"/>
    </inertial>
  </link>
</robot>
```

And here the SRDF file created during MoveIt! configuration package generation:

```
<?xml version="1.0" ?>

<robot name="robot">

    <group name="m">
        <chain base_link="base" tip_link="lever_arm" />
    </group>
    <group name="e">
        <chain base_link="lever_arm" tip_link="pointer" />
    </group>

    <group_state name="hm" group="m">
        <joint name="q0" value="0" />
        <joint name="q1" value="0" />
    </group_state>
    <group_state name="he" group="e">
        <joint name="q2" value="0" />
    </group_state>

    <end_effector name="ee" parent_link="lever_arm" group="e" parent_group="m" />

    <virtual_joint name="virtual_joint" type="fixed" parent_frame="world" child_link="footprint" />

    <disable_collisions link1="base" link2="lever_arm" reason="Never" />
    <disable_collisions link1="base" link2="pointer" reason="Never" />
    <disable_collisions link1="base" link2="rotatory" reason="Adjacent" />
    <disable_collisions link1="lever_arm" link2="pointer" reason="Adjacent" />
    <disable_collisions link1="lever_arm" link2="rotatory" reason="Adjacent" />
    <disable_collisions link1="pointer" link2="rotatory" reason="Never" />
</robot>
```

TF frames seem ok

I already have tried to find an answer on [answers.ros.org](https://answers.ros.org/question/275893/moveit-rviz-motionplanning-plugin-broken-robot-geometry/)

### Your environment
* ROS Distro: Lunar
* OS Version: Ubuntu 17.04

### Steps to reproduce

To generate the  Moveit! Configuration I am performing the following steps:

1. Click "Create New MoveIt Configuration Package"
2. Enter path to URDF-File and click "Load Files" (The manipulator geometry is displayed on the right of the screen, according to my configuration in the URDF file)
3. Click "Self-Collisions" and "Generate Collision Matrix"
4. Click "Virtual Joints" and "Add Virtual Joint" with following parameters: Virtual Joint Name: virtual_joint; Child Link: footprint; Parent Frame Name: world; Joint Type: fixed
5. Click "Save"
6. Click "Planning Groups" and "Add Group" with the following parameters: Group Name: m Kinematic Solver: None; defaults
7. Click "Add kin. Chain" and select "base" as Base Link and "lever_arm" as Tip Link
8. Click "Save"
9. Perform Steps 6 to 8 with parameters: Group Name: e Kinematic Solver: None; defaults and select "lever_arm" as Base Link and "pointer" as Tip Link
10. Click "Robot Poses" and "Add Pose" with parameters: Pose Name: hm Planning Group: m; q0=0; q1=0
11. Click "Save"
12. Click "Robot Poses" and "Add Pose" with parameters: Pose Name: he Planning Group: m; q2=0
13. Click "Save"
14 Click "End Effectors" and "Add End Effectors" with parameters: End Effector Name: ee; End Effector Group: e; Parent Link: lever_arm; Parent Group: m
15. Click "Save"
16. Skip "Passive Joints"
17. Enter author information
18. Click "Configuration Files", Enter path, click "Generate Package" and "Exit Setup Assistant"

Rebuild workspace with `catkin_make`
Source /opt/ros/lunar/setup.zsh and ...catkin_ws/devel/setup.zsh
run `roslaunch moveit_test demo.launch`

### Expected behaviour
I am pretty new to MoveIt!, but I could not get any help. I would have expected to see results like in the tutorials.

### Actual behaviour
The manipulator geometry collapses.

### Backtrace or Console output

[Console Output of RViz](https://gist.github.com/dbasa/16c026b01d09084ceab05cb02e59de27)

dbasa, 2017-11-16:
The missing visuals for primitives might be linked to issue [1151 at ros-visualization](https://github.com/ros-visualization/rviz/issues/1151)

gavanderhoorn, 2017-11-16:
@dbasa wrote:
> I already have tried to find an answer on answers.ros.org

indeed. I've added some comments there.

Could you please not cross-post issues like this? It could lead to disjoint discussions with the potential for duplication of effort which wastes everyone's time.

dbasa, 2017-11-20:
OK, found a bug in urdfdom_headers. When parsing the URDF file std::stod is used to parse the floating point numbers. But, std::stod is locale dependent. My system is using german locale, so std::stod is looking for a comma instead of looking for a dot as decimal separator. When parsing pi/2 (1.57079632679) std::stod returns just 1.
Going to close this.

closed 2017-11-20 by dbasa

gavanderhoorn, 2017-11-20:
Are you planning to report this over at [ros/urdfdom_headers/issues](https://github.com/ros/urdfdom_headers/issues)?

dbasa, 2017-11-20:
Yes I am!

gavanderhoorn, 2017-11-20:
If/when you do, please comment here with a link to the issue you posted.

dbasa, 2017-11-20:
Reported this issue over at [ros/urdfdom_headers/issues](https://github.com/ros/urdfdom_headers/issues/41)