moveit/moveit #694: [kinetic] Invalid Trajectory: start point deviates from current robot state more than 0.01

Vonstafenberg, 2017-11-21:
Hi,

I am using Moveit with ROS kinetic for planning the motions for a UR5 arm. I am using the Track_IK kinematic solver. The problem is , I am speicifying simple goal poses for the gripper in cartesian space, but I am getting the error message in the terminal of moveit_planning execution.launch.

> **[ERROR] [1511284205.660285187]: 
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'arm_elbow_joint': expected: 0, current: 1.84494
[ INFO] [1511284205.660356791]: Execution completed: ABORTED
[ INFO] [1511284205.700050212]: Execution request received for ExecuteTrajectory action.
[ WARN] [1511284205.708508088]: Controller  failed with error code INVALID_GOAL
[ WARN] [1511284205.708629940]: Controller handle  reports status FAILED
[ INFO] [1511284205.708804955]: Execution completed: FAILED
[ INFO] [1511284205.739949934]: Execution request received for ExecuteTrajectory action.**

I looked into the issue mentioned [here](https://github.com/ros-industrial-consortium/descartes/issues/196). But I ma using trac_Ik instead of Descartes. So I am not sure how to implement a proper solution.

I also referenced [this](https://groups.google.com/forum/#!topic/moveit-users/x5K6lVuiGqM), but then changing the

>  **/move_group/trajectory_execution/allowed_start_tolerance** to 0.0 is giving me another error.

> **[ WARN] [1511285456.138267775]: Controller  failed with error code INVALID_GOAL
[ WARN] [1511285456.138359326]: Controller handle  reports status FAILED
[ INFO] [1511285456.138473683]: Execution completed: FAILED
[ INFO] [1511285456.168054862]: Execution request received for ExecuteTrajectory action.**

in the terminal where I launch the UR_driver to interface with the real robot I have the error

> **[ERROR] [1511285455.959112929]: Goal start doesn't match current pose
[ INFO] [1511285455.995846536]: on_goal
[ERROR] [1511285455.996445868]: Goal start doesn't match current pose
[ INFO] [1511285456.036065982]: on_goal**

and in the terminal where I launch the node I get 

> **[ INFO] [1511285455.930223750]: ABORTED: Solution found but controller failed during execution
[ INFO] [1511285455.968737779]: ABORTED: Solution found but controller failed during execution
[ INFO] [1511285456.009758052]: ABORTED: Solution found but controller failed during execution
[ INFO] [1511285456.049777660]: ABORTED: Solution found but controller failed during execution
**

I am also attaching part of the code here to get a better view of the things

` int main(int argc, char **argv)
  {
    ros::init(argc, argv, "move_group_pick");
    ros::NodeHandle node_handle;
    ros::AsyncSpinner spinner(1);
    spinner.start();
    ros::Duration sleep_time(3.0);
    group.reset( new moveit::planning_interface::MoveGroupInterface ("arm"));
    planning_scene_interface.reset(new moveit::planning_interface::PlanningSceneInterface());
    transformListener = boost::shared_ptr<tf::TransformListener>(new tf::TransformListener(node_handle, 
    ros::Duration(1.0)));
   display_publisher = node_handle.advertise<moveit_msgs::DisplayTrajectory>("/move_group/display_planned_path", 1, true);
    rviz_updater = node_handle.advertise<std_msgs::Empty>("/rviz/moveit/update_goal_state", 1, true);
    timer = node_handle.createTimer(ros::Duration(0.1), timerCallback);
    ROS_INFO("Reference frame: %s", group->getPlanningFrame().c_str());
    ROS_INFO("Reference frame: %s", group->getEndEffectorLink().c_str());
    client = node_handle.serviceClient<ur_msgs::SetIO>("ur_driver/set_io");
    srand(time(NULL));
    target_pose0 = Pose_pick_up();
    if (move2pose(group, target_pose0) == -1) return (-1) ;
    sleep_time.sleep();	
    target_pose0 = Pose_mrkr();
    if (move2pose(group, target_pose0) == -1) return (-1) ;
    sleep_time.sleep();
    /// spawn/attach the box here
   ros::Publisher planning_scene_diff_publisher = node_handle.advertise<moveit_msgs::PlanningScene>("planning_scene", 1);
   while(planning_scene_diff_publisher.getNumSubscribers() < 1)
   {
     ros::WallDuration sleep_t(0.5);
     sleep_t.sleep();
   }

   moveit_msgs::AttachedCollisionObject attached_object;
   attached_object.link_name = "gripper_link";
   attached_object.object.header.frame_id = "gripper_link";      
   attached_object.object.id = "box";   
}
///----------.......................................
//.................................`
Any help is much appreciated.
@v4hn , @gavanderhoorn

BryceStevenWilley, 2017-11-21:
> I looked into the issue mentioned here. But I'm using trac_Ik instead of Descartes. So I am not sure how to implement a proper solution.

From that thread, it seems like the general solution is independent of whatever IK solver you're using, Trak-IK or Descartes. I think the most relevant part of that thread is this:

> The way to handle this is to "connect" the start of the descartes-generated trajectory with the current position of the robot. You can do this manually by inserting a trajectory_msgs::JointTrajectoryPoint with the positions variables set to the robot's current position into the trajectory before attempting execution.

You also mentioned that:

> I also referenced this, but then changing the

>    /move_group/trajectory_execution/allowed_start_tolerance to 0.0 is giving me another error.
 >   [ WARN] [1511285456.138267775]: Controller failed with error code INVALID_GOAL
 >   [ WARN] [1511285456.138359326]: Controller handle reports status FAILED
>    [ INFO] [1511285456.138473683]: Execution completed: FAILED
 >   [ INFO] [1511285456.168054862]: Execution request received for ExecuteTrajectory action.

> in the terminal where I launch the UR_driver to interface with the real robot I have the error

   > [ERROR] [1511285455.959112929]: Goal start doesn't match current pose
>    [ INFO] [1511285455.995846536]: on_goal
>    [ERROR] [1511285455.996445868]: Goal start doesn't match current pose
 >   [ INFO] [1511285456.036065982]: on_goal

This is because the controller on the URs have a similar protection mechanism that you disabled in MoveIt: it will refuse to execute the trajectory if the start state isn't the current state of the robot. (See [this](https://github.com/ThomasTimm/ur_modern_driver/issues/35) for some more info). I would suggest re-enabling the check in MoveIt and adding your current state to the beginning of the path you are trying to execute.

It looks like the relevant portion of your code is in the `move2pose` method.

Hope that helps you figure out how to tackle your problem.

Vonstafenberg, 2017-11-22:
@BryceStevenWilley 

Thank you for the reply.

> This is because the controller on the URs have a similar protection mechanism that you disabled in MoveIt: it will refuse to execute the trajectory if the start state isn't the current state of the robot

Ok so this is the problem.

>  I would suggest re-enabling the check in MoveIt and adding your current state to the beginning of the path you are trying to execute.

I would check this and see if it solves the problem. and yes I am using the move2pose method. I am trying simple poses and yet I am facing problems.

v4hn, 2017-11-23:
> joint 'arm_elbow_joint': expected: 0, current: 1.84494

This means your planned trajectory starts at `arm_elbow_joint` at position `0`, but the actual arm is in position `1.84494`. That's a difference of more than 45deg!
So apparently you plan your trajectory from an incorrect start state (I guess all 0s?).
As @BryceStevenWilley already said the problem is in - or related to - your `move2pose` method.

I'll close this thread. It's not a bug in moveit.

closed 2017-11-23 by v4hn

Privilger, 2020-06-20:
Hi, 

I met the same error. But I think my case can be different. My error is shown bellow:

```
[ERROR] [1592645990.216860215]: 
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'wrist_roll_joint': expected: 3.14087, current: -3.1404
```

I think it is another error that due to the rotation expression using roll, pitch, and yaw. The `wrist_roll_joint` is in a place that it sometimes is measured as ` -3.1404` and sometimes `3.14087`. The measurement noise makes it happend. I think maybe using Quaternion to express the rotation can help avoiding this problem.

rhaschke, 2020-06-20:
A switch from -3.14 to +3.14 should only happen in case of Euler angles or the like. A _real_ robot joint should measure its joint angle in a more reliable fashion. What kind of robot you are using?

Privilger, 2020-06-20:
Yes, I agree it is the problem of using Euler angles. 
I'm using the Fetch Mobile Manipulator. And it is a real robot, not the Gazebo simulation robot.

v4hn, 2020-06-21:
Quaternions do not solve anything here, as the joint state `-2pi` and `0` are quite different things if the joint has position limits.

> > A _real_ robot joint should measure its joint angle in a more reliable fashion.
> 
> I'm using the Fetch Mobile Manipulator.

Are there unlimited joints in the fetch? Didn't have the chance to work with one myself.

The PR2 simply keeps going, without normalizing joints values.
For its endless joints you can get joint state readings 2pi,10pi,50pi...
It looks like fetch implicitly normalizes?

Please try #2157 , it should fix your issue.

Privilger, 2020-06-22:
That looks good to me!

It seems the bug from the source code.
I use apt to download MoveIt, not from the source code. Should I just wait for a new release?

Thanks a lot!

v4hn, 2020-06-22:
> I use apt to download MoveIt, not from the source code. Should I just wait for a new release?

That might take a while. So if you need the fix right away, compile MoveIt from source (if you run kinetic, you just need to compile the `kinetic-devel` branch, everything else is already installed) and add the patch on top (it's not merged and backported yet).