moveit/moveit #697: Errors when build moveit from souce

XieLongZju, 2017-11-25:
### Description
Hello,

I try to build moveit from source, but keep getting errors.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source kinetic-devel

### Steps to reproduce
I followed this tutorial http://moveit.ros.org/install/source/

>rosdep update
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install python-wstool python-catkin-tools clang-format-3.8
mkdir -p ~/ws_moveit/src
cd ~/ws_moveit/src
wstool init .
wstool merge https://raw.githubusercontent.com/ros-planning/moveit/kinetic-devel/moveit.rosinstall
wstool update
rosdep install -y --from-paths . --ignore-src --rosdistro kinetic
cd ..
catkin config --extend /opt/ros/kinetic --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build

### Expected behaviour
Build success!

### Actual behaviour

 /opt/ros/kinetic/lib/libresource_retriever.so:_ undefined reference to `curl_easy_setopt@CURL_OPENSSL_3'

 /opt/ros/kinetic/lib/libresource_retriever.so: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_3'

/opt/ros/kinetic/lib/libresource_retriever.so: undefined reference to `curl_easy_init@CURL_OPENSSL_3'

/opt/ros/kinetic/lib/libresource_retriever.so: undefined reference to `curl_global_cleanup@CURL_OPENSSL_3'

/opt/ros/kinetic/lib/libresource_retriever.so: undefined reference to `curl_easy_perform@CURL_OPENSSL_3'

/opt/ros/kinetic/lib/libresource_retriever.so: undefined reference to `curl_global_init@CURL_OPENSSL_3'

collect2: error: ld returned 1 exit status
make[2]: *** [devel/lib/moveit_setup_assistant/collisions_updater] Error 1
make[1]: *** [CMakeFiles/moveit_setup_assistant_updater.dir/all] Error 2
make: *** [all] Error 2

hjzh4, 2018-07-06:
Hi I met the same issue, have you got it work?

rhaschke, 2018-07-06:
Looks like an issue in dependencies of resource_retriever. I suspect that you are missing libcurl.so.
Could you please check the following command:
```bash
$ ldd /opt/ros/kinetic/lib/libresource_retriever.so | grep curl
	libcurl.so.4 => /usr/lib/x86_64-linux-gnu/libcurl.so.4 (0x00007fc40b78b000)
```
It should report (as in my case) the presence of libcurl.so.

hjzh4, 2018-07-06:
Hi, I got this output:
```
libcurl.so.4 => /usr/lib/x86_64-linux-gnu/libcurl.so.4 (0x00007f52ff7ee000)
```
I already installed moveit via apt. 
Anyway, thanks for your help!

rhaschke, 2018-07-06:
@hjzh4 I guess, you've got this output _after_ installing moveit via apt, didn't you?

hjzh4, 2018-07-06:
I don’t think so, checked it before🤣

rhaschke, 2018-07-06:
Do you still get the error when building MoveIt from source?
Do you have a non-system version of curl installed somewhere?

hjzh4, 2018-08-25:
Yay, I think the problem is curl-gnutls v.s. curl-openssl version. However the ros and gazebo I'm using are all dependent on curl-openssl.

amanjhunjhunwala, 2018-08-27:
+1 same error

arrfou90, 2018-09-17:
I had the same issue   (it is a dependency issue: curl-openssl version).  I solved it by removing curl-openssl that was installed inside Anaconda2.

tiexuedanxin, 2018-11-02:
hey,i get the output "libcurl.so.4 => /usr/lib/x86_64-linux-gnu/libcurl.so.4 (0x00007f6ae7650000)",
but when i continue to catkin build,it shows the same mistakes as same,so how to solve the problem,is there someone could help me,thanks for advance

rhaschke, 2018-11-02:
Probably you have the includes found from another curl installation. As stated above, please check your (multiple) curl installations.

tiexuedanxin, 2018-11-02:
> Probably you have the includes found from another curl installation. As stated above, please check your (multiple) curl installations.

could you tell me your email address,i have a email to you

rhaschke, 2018-11-03:
@tiexuedanxin As I said multiple times, this error is related to _your_ particular system. We cannot help you on this. Try to do a clean install. As you can see [here](https://travis-ci.org/ros-planning/moveit/branches), MoveIt builds from source without any issue - on a clean system.

closed 2018-11-03 by rhaschke

tiexuedanxin, 2018-11-04:
> @tiexuedanxin As I said multiple times, this error is related to _your_ particular system. We cannot help you on this. Try to do a clean install. As you can see [here](https://travis-ci.org/ros-planning/moveit/branches), MoveIt builds from source without any issue - on a clean system.
Do you means to reinstall the ubuntu system?!

YuhengZhi, 2020-09-06:
I am having a similar problem. I am building Moveit from source on Ubuntu 20.04, which has to use ROS distro Noetic. The errors I got are almost the same as the original description except that the references missing have a suffix of CURL_OPENSSL_4 instead of CURL_OPENSSL_3. A possible clue:
```
$ readelf -Ws /opt/ros/noetic/lib/libresource_retriever.so | grep curl_global_cleanup
5: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND curl_global_cleanup@CURL_OPENSSL_4 (3)
$ readelf -Ws /lib/x86_64-linux-gnu/libcurl.so.4 | grep curl_global_cleanup
512: 0000000000030b40    55 FUNC    GLOBAL DEFAULT   15 curl_global_cleanup@@CURL_OPENSSL_4
```
note how the referee and referrer are different by the number of '@'s.

rhaschke, 2020-09-07:
Sorry, we cannot help on this issue: it's a problem with your system installation (of `libcurl`). It works for us on a clean Ubuntu system install as Travis CI proofs.

dnandha, 2021-02-04:
> I had the same issue (it is a dependency issue: curl-openssl version). I solved it by removing curl-openssl that was installed inside Anaconda2.

Thanks @arrfou90! After deactivating the conda base environment it worked!

ranjitkathiriya, 2021-02-10:
Thanks, @arrfou90!!!

ha7ilm, 2024-03-17:
> This is probably because some ROS packages are built under conda environment. And the required python interpreter by the ROS packages are specified to the python interpreter under conda directory. And in this case, some librarys might be misled to the libs under conda directory.

Actually, exiting the conda environment (with Ctrl+D or by closing the terminal, until (base) or your appropriate environment name is not visible) and cleaning the project with `catkin clean -b` did the trick!

amartynyuk, 2025-09-07:
Same issue, solved by removing libcurl from conda:
`conda uninstall libcurl`