moveit/moveit #702: Add official support for CHOMP

bmagyar, 2017-12-02:
With #626 CHOMP should back on it's feet. On our last maintainer meeting it was proposed to make CHOMP supported in the Moveit Setup Assistant as well. 
After this it could be added back to the released packages.

@simonschmeisser , do you have anything else you would like to add? 

FYI:  @IanTheEngineer @jrgnicho

davetcoleman, 2018-05-10:
+1 to this. Perhaps we should target this goal at end of summer after one of the GSoC students puts some additional work into it.

bmagyar, 2018-05-11:
Big +1. To ensure some stability of general features and keep duplication
low it'd be really nice to have a shared set of high level tests we could
set up for every planner.

simonschmeisser, 2018-05-11:
We could turn it into a benchmark to ignite some competition ;)

davetcoleman, 2018-05-17:
@simonschmeisser can you put together a checklist of outstanding issues in your first comment in this issue? add links to other issues in this repo for reference. this will help you, @raghavendersahdev, and others divider and conquer this task

@raghavendersahdev I just audited the [tutorial](http://tutorials.picknik.ai/doc/chomp_interface/chomp_interface_tutorial.html) and see some checklist items there:
- [ ] Link the text "work in progress" to this issue to track progress
- [ ] Rename tutorial title to "CHOMP Planner"
- [ ] Add explanation of why moveit must be installed from source (because chomp is not released yet)
- [ ] Turn the "Note:" callout into a properly formatted blue box note
- [ ] Steps 1-5 should be removed by adding this directly into the setup assistant template so that all robots automatically have these configurations
- [ ] Split up step 7 into two steps so it looks better
- [ ] Add section to bottom of tutorial explaining what you should expect to see different from OMPL plans
- [ ] Document some of the params that can be tweaked for CHOMP

raghavendersahdev, 2018-05-30:
Small update on this, Pending tasks currently:

- [ ] Turn the "Note:" callout into a properly formatted blue box note
- [ ]  Steps 1-5 should be removed by adding this directly into the setup assistant template so that all robots automatically have these configurations
- [ ] Add section to bottom of tutorial explaining what you should expect to see different from OMPL plans
- [ ] Document some of the params that can be tweaked for CHOMP

Will aim to address 3 of these 4 items by Thursday (May 31,2018) in a new PR. and try to do the setup assistant template one by the weekend. 

Others checklist items addressed in the pull request [PR#173.](https://github.com/ros-planning/moveit_tutorials/pull/173/)

bmagyar, 2018-05-30:
Just one note, you may make it an item if you think it's important:
AFAIK all moveit planners namespace (all or part of) their params by planning group. I think CHOMP should do the same.