moveit/moveit #705: MoveIt! Support for Multiple, Asynchronous Trajectories

rkeatin3, 2017-12-04:
I posted [this](https://groups.google.com/forum/#!topic/moveit-users/iNvIcy6hZNo) on the MoveIt! user-group but did not receive any input, and so I'll follow-up here: Does MoveIt! support executing multiple, asynchronous trajectories simultaneously?

I have been working in Jade and see that the `TrajectoryExecutionManager` is responsible for the behavior described in my post to the user-group: it cancels the currently executing trajectory whenever another comes in (despite that the group for each is different). And so it looks like changes to support the behavior I want would have to happen there. 

Has anyone tried to get something similar to work? Any suggestions? It's certainly possible that I'm doing something wrong as well.

rhaschke, 2017-12-05:
Executing two trajectories in parallel, even if they belong to different group names, isn't safe.
First, one would need to ensure, that the joint sets involved are disjoint (not only the group names).
Second, even if joint sets are disjoint, trajectories might collide with each other when executed in parallel.

How do you ensure safety?

rhaschke, 2017-12-05:
After reading your original post, I suggest to start two independent move_group nodes with different namespaces. Then you can run your planning and execution in parallel. However, that only works safely, if the robot's workspaces are disjoint as in your case.

rkeatin3, 2017-12-08:
I wasn't sure about the MoveIt methodology in terms of safety, but I understand now that the behavior I desire is explicitly forbidden because it would be unsafe in other cases. Thanks for the tip on how I might be able to accomplish it.

With that in mind, I believe that the current behavior isn't actually that safe: I was able to plan two arm motions (separately but with the same start state), partially execute one, and then execute the other. I believe that means that if my robots _had_ overlapping workspaces, that there could have been a collision. Does that mean when a trajectory is executed that it should check if all the joints of the robot are still at the start state (rather than just checking the joints that belong to the move group being controlled)?

rhaschke, 2017-12-08:
That's a good point. Currently, indeed the start-state validation only considers the joints that are part of the execution trajectory. Doing it differently, becomes difficult as the trajectory doesn't contain the other joints.
@v4hn @davetcoleman Do you have any ideas to fix this issue too?

cemcetin, 2019-07-10:
@rkeatin3 did you find any other method to do asynchronous planning for multiple groups that might be safer?

rkeatin3, 2019-07-16:
I did not. I ended up using two separate move_group nodes as far as I remember.

closed 2019-07-16 by rhaschke