moveit/moveit #709: Unclear how to combine MoveGroupCommander compute_cartesian_path and execute methods

broesdecat, 2017-12-05:
### Description

From the (limited) API documentation, I would have expected the following code to do the job:
```
plan = commander.compute_cartesian_path(waypoints, 0.01, 0.001, True)
commander.execute(plan.serialize(), True)
```
However, it results in an error on the serialization of a tuple:
```
Traceback (most recent call last):
	...
  File "appHandler.py", line 113, in execute
    commander.execute(plan, True)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 488, in execute
    return self._g.execute(conversions.msg_to_string(plan_msg))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/moveit_commander/conversions.py", line 43, in msg_to_string
    msg.serialize(buf)
AttributeError: 'tuple' object has no attribute 'serialize'
```
I guess there is some subtle thing I am missing here, but I can't derive it from any of the documentation or examples I found.

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 16.04
* Binary build

mbharatheesha, 2017-12-05:
As per the documentation, the call to `compute_cartesian_path` returns a tuple which contains the planned robot trajectory AND a fraction of how much of the planned path will be followed by the trajectory. However, the execute call only requires the robot trajectory and not the entire tuple. Hope this helps in addressing your issue.

keerthanamanivannan, 2018-01-16:
yes. @mbharatheesha is right. change your code to 
`plan, fraction = commander.compute_cartesian_path(waypoints, 0.01, 0.001, True)
commander.execute(plan, True)` 
now the robot will execute the plan.

felixvd, 2019-10-11:
I have also stumbled over the tutorials mentioning how to calculate the cartesian trajectory, but not how to execute it. Extending the tutorials with extra examples and explanations for this would be a great WMD2019 issue. Could someone tag it? @BryceStevenWilley @davetcoleman