moveit/moveit #710: moveit::core::RobotState::setVariablePositions API question

abishekh, 2017-12-05:
### Description
`void moveit::core::RobotState::setVariablePosition  (  const std::string & variable, double  value)`
`void moveit::core::RobotState::setVariablePosition  ( int index, double  value ) `

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source 

Is there any particular reason why there aren't  'const' versions of the API for the second argument (double value) ? I see the vector and array versions are const  `moveit::core::RobotState::setVariablePositions ( const std::vector< std::string > & variable_names, const std::vector< double > & variable_position) ` but I was wondering about setting individual values specifically.

rhaschke, 2017-12-05:
Usually it doesn't make sense to declare constness for arguments passed by value (i.e. creating a copy), because the function can do anything it likes with those args.
Arrays / vectors are passed by reference. Here the const declaration promises, that the function will not change those arguments.

abishekh, 2017-12-06:
thats exactly the functionality I'm looking for with setVariablePosition() .

rhaschke, 2017-12-06:
Which functionality you are exactly referring to? Please be more specific.
Is there anything you are still missing?

abishekh, 2017-12-06:
Functionality that I'm looking for : 

> API that promises that the function will not change the arguments. 

Especially when setting a joint position or constraint or similar.

rhaschke, 2017-12-07:
As I said above, all versions of `setVariablePosition()` will not change the passed joint position arguments.
Either because the argument is passed by value (i.e. copied for use in the function) or because it's passed by const reference / const pointer.

closed 2017-12-07 by abishekh